wadpac/GGIR #6: g.cwaread error in Windows

vincentvanhees, 2017-05-04:
After installing GGIR in Windows and trying to read a cwa file with g.cwaread using

`P = g.cwaread("D:/test.cwa",start=1,end=10)`

I get

`Error in Rcpp::sourceCpp("src/numUnpack.cpp") : 
  file not found: 'C:/Users/Vincent/Documents/src/numUnpack.cpp'
In addition: Warning messages:
1: In normalizePath(path.expand(path), winslash, mustWork) :
  path[1]="src": The system cannot find the file specified
2: In normalizePath(path.expand(path), winslash, mustWork) :
  path[1]="src/numUnpack.cpp": The system cannot find the path specified`

I suppose installation / compilation fails.

@Mirkes can you help please?

Mirkes, 2017-05-04:
Dear Vincent,

I do not have experience in work with Rcpp in packages. I'll try to understand. It looks like error with paths because I saw that R try to find cpp file in completely wrong directory. All other messages shows the same problem.

I hope that I found solution soon.

vincentvanhees, 2017-05-06:
Dear Evgeny, that would be very kind of you. Here is my progress so far:

The issue with the code above is that function [g.cwaread](https://github.com/wadpac/GGIR/blob/master/R/g.cwaread.R) has on line 44 and 45:
 Rcpp::sourceCpp('src/numUnpack.cpp')
 Rcpp::sourceCpp('src/resample.cpp')

These two lines are needed when working with the uncompiled package. However, when the package is compiled these calls are not meaningful anymore I suppose.

When I comment the sourceCpp calls out, re-compile, and re-install the package, and test g.cwaread again, I get:
`> P <- g.cwaread(fileName="test.cwa",start=1,end=10)`
`Error in .Call("GGIR_numUnpack", PACKAGE = "GGIR", pack) : 
  "GGIR_numUnpack" not available for .Call() for package "GGIR"'

Which sounds similar to the bug reported in another package on stackoverflow [here](http://stackoverflow.com/questions/22594238/rcpp-rcpp-package-skeletonmypackage-rcpp-hello-world-not-available-for-ca). The problem they had does not seem to apply to our situation and I have been carefully comparing our code with the code of other packages, but no luck so far.

vincentvanhees, 2017-05-06:
I now also realise that this problem is not Windows specific, but is the same in Linux (Ubuntu in my case)

Mirkes, 2017-05-06:
Dear Vincent,

Many thanks for your investigation. Now I have no idea but I will try to find solution.

Best regards,
Evgeny

Mirkes, 2017-05-06:
Dear Vincent,

Names of functions looks like strange: GGIR_numUnpack. Originally name of function was numUnpack. I am not sure, but may be necessary to change names of functions?

Best regards,
Evgeny

vincentvanhees, 2017-05-07:
I just started from scratch and reimplemented the cpp code using the Rcpp.package.skeleton function from the Rcpp package. I deleted the .h and .c files I generated previously and I have added documentation files for numUnpack and resample.  These changes are now in release 1.5-4: https://github.com/wadpac/GGIR/releases/tag/v1.5-4

vincentvanhees, 2017-05-07:
Problem not fixed yet:
The above changes allowed me to  build the package in Linux and it would have been fine half a year ago, but since earlier this year CRAN requires/recommends that native routines are registered. So, I would not be able to upload the package as such to CRAN. Next, I followed [R Writing Extensions](https://cran.r-project.org/doc/manuals/r-release/R-exts.html#Registering-native-routines) and added the init.c file, and updated the NAMESPACE file.

I am still  able to build the package, but when I call g.cwaread I get again the error message:
Error in .Call("GGIR_numUnpack", PACKAGE = "GGIR", pack) : 
  "GGIR_numUnpack" not available for .Call() for package "GGIR"

Note that the  files RcppExperts.cpp, and RcppExports.R and the GGIR_ addition to numUnpack are automatically generated by Rcpp::compileAttributes() when building the package, so I do not think that the problem is a typo in that file.

vincentvanhees, 2017-05-08:
I have posted a summary of this issue on [stackoverflow](http://stackoverflow.com/questions/43846017/r-package-with-rcpp-builds-but-when-calling-function-error-in-call). Hopefully someone there can spot the mistake.

vincentvanhees, 2017-05-08:
New release, which hopefully has fixed the issue: https://github.com/wadpac/GGIR/releases/tag/v.1.5-5

vincentvanhees, 2017-05-09:
I have also added the windows binary too: https://github.com/wadpac/GGIR/releases/download/v.1.5-5/GGIR_1.5-5.zip
To me this seems to work, if you still get errors then restart R.

closed 2017-05-09 by vincentvanhees