wadpac/GGIR #7: Part 2 error

DiegoGBassani, 2017-05-10:
Hi Vincent - Thanks for the most recent update. I am now running it again and getting a new type of error when it reaches part2:

Error in matrix(0, nwindow_f, 1) : invalid 'nrow' value (< 0)

I have now set all metrics to == TRUE so I do not run into the same problem as before, but still getting this nrow error.

Thanks

Diego

vincentvanhees, 2017-05-10:
Hi Diego, I meant to write = TRUE, not == TRUE, in our [google group discussion](https://groups.google.com/d/topic/rpackageggir/RY8wzp7_Dos/discussion). If that does not fix it, please do some research first: Start with the default settings - which is the same as not entering any function arguments - and then itereratively adding your desired input argument. Thanks

DiegoGBassani, 2017-05-10:
Thanks - and yes, I am using =TRUE - that is not the issue. Let ,me try as you suggested and see how it goes.
D

This e-mail may contain confidential, personal and/or health information(information which may be subject to legal restrictions on use, retention and/or disclosure) for the sole use of the intended recipient. Any review or distribution by anyone other than the person for whom it was originally intended is strictly prohibited. If you have received this e-mail in error, please contact the sender and delete all copies.

closed 2017-07-22 by vincentvanhees