wadpac/GGIR #13: GGIR version 1.5-7 bug

JianWang2016, 2017-05-17:
Dear Vincent,

Per your instruction on forum at https://groups.google.com/forum/#!topic/RpackageGGIR/kjJMePHDU_Q,
I am submitting an issue here.

The bug of src/numUnpack.cpp is NOT fixed in g.inspectfile(), g.getmeta(), g.calibrate() and a number of other places using the above functions.

Your attention is greatly appreciated.

> library(GGIR)
> I = g.inspectfile(datafile)
Error in Rcpp::sourceCpp("src/numUnpack.cpp") : 
  file not found: 'src/numUnpack.cpp'
In addition: Warning messages:
1: In normalizePath(dirname(file)) :
  path[1]="src": No such file or directory
2: In normalizePath(file, winslash = "/") :
  path[1]="src/numUnpack.cpp": No such file or directory
> I = g.getmeta(datafile)
Error in Rcpp::sourceCpp("src/numUnpack.cpp") : 
  file not found: 'src/numUnpack.cpp'
> I = g.calibrate(datafile)
Error in Rcpp::sourceCpp("src/numUnpack.cpp") : 
  file not found: 'src/numUnpack.cpp'

vincentvanhees, 2017-05-17:
thanks Jian, I am aware of this and it will be fixed in the next release, which I aim to upload later this week

vincentvanhees, 2017-05-17:
Now in release 1.5-8 here on github https://github.com/wadpac/GGIR/releases/tag/1.5-8
I will upload this to CRAN later this week

JianWang2016, 2017-05-17:
Thanks so much Vincent for getting a new release so quickly.  
It looks great so far!

JianWang2016, 2017-05-17:
Vincent,

I just started to read in a 2 week .cwa file using the new version.  From the g.inspectfile, blocks=991997, that is the number of blocks I have?
It is about 10 min running
g.shell.GGIR(mode=c(1,2),datadir=datadir,outputdir=outputdir,studyname=studyname, f0=1,f1=1,
              do.report=c(2),overwrite=FALSE,visualreport=TRUE,viewingwindow=1)

Does the output look right so far for you?  With these speed, it is going to take 10 million min to load all the blocks?  

> I = g.inspectfile(datafile)
> I
$header
                      value
uniqueSerialCode      31999
frequency               100
start            1493411438
device              Axivity
firmwareVersion          44
blocks               991997

$monc
[1] 4

$monn
[1] "axivity"

$dformc
[1] 4

$dformn
[1] "cwa"

$sf
[1] 100

$filename
[1] "31999_0000000204.cwa"

> g.shell.GGIR(mode=c(1,2),datadir=datadir,outputdir=outputdir,studyname=studyname, f0=1,f1=1,
+               do.report=c(2),overwrite=FALSE,visualreport=TRUE,viewingwindow=1)

   g.shell.GGIR {GGIR} by Vincent van Hees

________________________________________________________________________________________________________________________
Part 1
[1] "P1 file1"
[1] "---------------------------------------------"
[1] "investigate calibration of the sensors..."
[1] "Loading block: 1"
[1] "Loading block: 2"
[1] "Loading block: 3"
[1] "Loading block: 4"
[1] "Loading block: 5"

End of file reached

stop reading because there is not enough data in this block
[1] "get meta data..."

warning: g.cwaread in beta version

Loading block: 1
Loading block: 2
Loading block: 3
Loading block: 4
Loading block: 5
Loading block: 6
Loading block: 7
Loading block: 8

vincentvanhees, 2017-05-17:
Looks good so far. The 991997 blocks are not the same 'blocks' as printed in the console. It is a bit ugly and confusing i have to admit, but g.cwaread defines a block in the data as a segment of 300 samples (3 seconds at 100Hz), while the block numbers other parts of GGIR are printing in your console are the blocks of data loaded, with about 12 hours of measurement per block for the autocalibration part and 24 hours per block in the get meta data part. So, this should not take much more than 15 minutes or so, depending of course on sample frequency and measurement duration.

vincentvanhees, 2017-05-17:
maybe I should rename the latter 'blocks' to 'chunks', because there is a function argument in GGIR called `chunksize` which relates to these larger blocks. In that way I can reserve the terms pages and blocks for everything that happens at the level of reading in the data from the cwa or bin files, and reserve the term chunks for the process of putting large 'chunks' of data in memory to process them. Related terms like window,  segment, period, epoch, and bout are already used elsewhere in the code. I will create a seperate issue for it now.

JianWang2016, 2017-05-17:
That makes a lot of sense.  Great!

I think there is another issue here.  As you can see, the ouputdir is not picked up in part2.  It also doesn't recognize metadatadir as input parameter.  The files from part1 is indeed stored in the outputdir as specified but part2 has its own idea where the meta data it reads from ;-)).

g.shell.GGIR(mode=c(1,2),
+              datadir="/lillyce/qa/digitalbiomarker/d1pam2/output/ax3TestDataCwa2",
+              outputdir="/lillyce/qa/digitalbiomarker/d1pam2/output/ax3TestAnalysis",
+              studyname=studyname, f0=1,f1=1,
+             do.report=c(2),overwrite=FALSE,visualreport=TRUE,viewingwindow=1)

   g.shell.GGIR {GGIR} by Vincent van Hees

________________________________________________________________________________________________________________________
Part 1

________________________________________________________________________________________________________________________
Part 2
 1
Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
In addition: Warning message:
In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file '/lillyce/qa/digitalbiomarker/d1pam2/output_ax3TestDataCwa2/meta/basic/NA', probable reason 'No such file or directory'

vincentvanhees, 2017-05-17:
My guess for now is that the code struggles with the folder name 'output' because GGIR creases a folder names 'output' and later on searches for it... I would have to investigate this as it would be nice if you could use any folder name you like.

As a quick fix, can you rename the argument for outputdir to:
outputdir = "/lillyce/qa/digitalbiomarker/d1pam2"

Next, your part2 output should be stored as subfolder inside:
'/lillyce/qa/digitalbiomarker/d1pam2/output_ax3TestDataCwa2'

JianWang2016, 2017-05-18:
This strategy works perfectly!  Thanks much.

JianWang2016, 2017-05-18:
Could you please do me a favor to remove the folder names in the comment above ("My guess ...) ?  I shouldn't have put our folder structure there.  Thanks much.

vincentvanhees, 2017-05-18:
Ok, I updated the folder names. Did you delete your own comment? It is not going to be very helpful for others to understand this conversation if you remove your own comments

closed 2017-05-18 by vincentvanhees

JianWang2016, 2017-05-19:
Vincent, I am so sorry.  
Here is the summary of what I reported:

The phase "output" is contained in my outputdir path, leading to confusion of go.part1.  As a result, contents which are supposed to be stored within a folder under outputdir ends up in a same named folder under datadir.  The program then complains the folder it needs not there. 

The workaround is to avoid using "output" in the outputdir path.

Thanks much Vincent.