wadpac/GGIR #18: Date not recognized in Actigraph input file using d-M-yyyy format

lvanwel, 2017-05-23:
g.getmeta produces the following error when using an ActiGraph input file with the d-M-yyyy date format (created by ActiLife v6.13.3 Firmware v2.4.0):

Loading block: 1
date not recognized

Loading block: 2
Loading block: 3
End of file reached
Error in seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) *  : 
  'from' cannot be NA, NaN or infinite

Input file line 4 reads: "Start Date 7-3-2016". When changing it manually to "Start Date 7/3/2016" the date is recognized and g.getmeta functions without issue.

vincentvanhees, 2017-06-05:
Thanks, I just fixed this and pushed changes to the master branch: [link to commit](https://github.com/wadpac/GGIR/commit/9c99d204d4729080bf74ee6ff335f44eeca7e08d)

closed 2017-06-05 by vincentvanhees

vincentvanhees, 2017-06-05:
and now also in release 1.5-10: https://github.com/wadpac/GGIR/releases/tag/v1.5-10