wadpac/GGIR #21: Part 4 Error in `[.data.frame`(S, , coln1 + ((i - 1) * 2) + 1) :    undefined columns selected

DiegoGBassani, 2017-06-06:
Hi Vincent - I have been able to make it work, finally, but it seems that now when part 4 tries to use the sleep log info I get an error in the data frame. I tried removing column names from first row, filling empty cells with '.' but I continue to get the same error. 

Here is the script I am using, and also an example of a sleep log file (.csv).

Any help will be much appreciated. 

Thanks
Diego

## Not run: 
mode= c(1,2,3,4)
datadir="/Volumes/G-DRIVEMINI/Sleep Files/Act_sleep_3m"
outputdir="/Volumes/G-DRIVEMINI/Sleep Files/Act_sleep_3m"
studyname="test3"
overwrite=TRUE
f0 = 1 
f1 = 5
desiredtz ="America/Sao_Paulo"
g.shell.GGIR(#-------------------------------
             # General parameters
             #-------------------------------
             mode=mode, 
             datadir=datadir, 
             outputdir=outputdir, 
             studyname=studyname, 
             f0=f0, 
             f1=f1, 
             overwrite = TRUE, 
             do.imp=TRUE,
             idloc=1, 
             print.filename=TRUE,
             storefolderstructure = FALSE,
             #-------------------------------
             # Part 1 parameters:
             #-------------------------------
             windowsizes = c(5,900,3600),
             do.cal=TRUE, 
             do.enmo = TRUE,
             do.lfenmo = TRUE,
             do.enmoa = TRUE, # Absolute taken from the Euclidean norm minus one.
             do.anglez=TRUE,
             do.en = TRUE,
             do.bfen = TRUE,
             do.hfen = TRUE,
             do.hfenplus = TRUE,         
             chunksize=1,
             printsummary=TRUE,
             #-------------------------------
             # Part 2 parameters:
             #-------------------------------
             strategy = 1,
             ndayswindow=7,
             hrs.del.start = 0,
             hrs.del.end = 0, 
             maxdur = 5,
             includedaycrit = 16, 
             L5M5window = c(0,24),
             M5L5res = 10,
             winhr = c(5,10),
             qlevels = c(0.1, 0.25, 0.5, 0.75, 0.9),
             qwindow=c(0,24), 
             ilevels = c(seq(0,400,by=50),8000), 
             mvpathreshold =c(100,120),
             epochvalues2csv = TRUE,
             IVIS_windowsize_minutes = 60, 
             IVIS_epochsize_seconds = 5,
             #-------------------------------
             # Part 3 parameters:
             #-------------------------------
             timethreshold= c(5,10),
             anglethreshold=3,
             ignorenonwear = TRUE,
             #-------------------------------
             # Part 4 parameters:
             #-------------------------------
             #Labels detected sustained inactivity periods by g.part3 as either nocturnal sleep or daytime sustained inactivity
             #Description
             #Loads output from g.part3 as stored in milestone data and sleep log information (if available) and 
             #then uses these information sources to define nocturnal sleep and daytime sustained inactivity.
             
             idloc=2,
             excludefirstlast = FALSE,
             includenightcrit = 4, #Minimum number of valid hours per night (24 hour window between noon and noon)
             def.noc.sleep = c(19,9),
             loglocation= "/Volumes/G-DRIVEMINI/Sleep Files/Act_sleep_3m/sleeplog3m_r.csv",
             outliers.only = FALSE,
             criterror = 6,
             relyonsleeplog = FALSE,
             #relyonsleeplog: If TRUE then sleep onset and waking time are defined based on timestamps derived from sleep log 
             #if FALSE (default) the sleep log is only used to guide the accelerometer-based detection. 
             #If participants were instructed NOT to wear the accelerometer during waking hours then set to TRUE, in all other scenarios set to FALSE (FALSE).
             
             sleeplogidnum = TRUE,
             colid=1, #Column number in the sleep log spreadsheet in which the participant ID code is stored (default = 1)
             coln1=2, #Column number in the sleep log spreadsheet where the onset of the first night starts
             
             do.visual = TRUE,
             #nnights=5, #Number of nights for which sleep log information should be available.
             #It assumes that this is constant within a study. If sleep log information is missing for certain nights then leave these blank
             
             #-------------------------------
             # Part 5 parameters:
             #-------------------------------
             # Key functions: Merging physical activity with sleep analyses
             threshold.lig = c(30,40,50),
             threshold.mod = c(100,120),
             threshold.vig = c(400,500),
             boutcriter = 0.8,
             boutcriter.in = 0.9,
             boutcriter.lig = 0.8,
             boutcriter.mvpa = 0.8,
             boutdur.in = c(10,20,30),
             boutdur.lig = c(1,5,10),
             boutdur.mvpa = c(1,5,10),
             timewindow = c("MM","WW"),
             #-----------------------------------
             # Report generation
             #-------------------------------
             do.report=c(4))

## End(Not run)

[sleeplog3m_r.xlsx](https://github.com/wadpac/GGIR/files/1055382/sleeplog3m_r.xlsx)

vincentvanhees, 2017-06-20:
Hi Diego,
Can you remind me what preceded your question? Does your question above relate to [this](https://groups.google.com/d/topic/rpackageggir/RY8wzp7_Dos/discussion) conversation by any chance? I am trying to help many people and there even is more than one Diego, by which I tend to forget conversations. 

What error message did you get?
Did you see [this](https://groups.google.com/d/topic/rpackageggir/0Ao8P9W7CN4/discussion) post?
Note that you should leave cells with missing values empty and not fill them with dots.
Cheers, Vincent

DiegoGBassani, 2017-06-20:
Hi Vincent - same Diego (I think...). I was able to make it work after deleting the 5th night (for which we have very little data) but the team in Pelotas is having issues running it in the full cohort still - they get the same error I described (column identification issue) even though the files work for me. We really do not know what to do... any clues?

This e-mail may contain confidential, personal and/or health information(information which may be subject to legal restrictions on use, retention and/or disclosure) for the sole use of the intended recipient. Any review or distribution by anyone other than the person for whom it was originally intended is strictly prohibited. If you have received this e-mail in error, please contact the sender and delete all copies.

vincentvanhees, 2017-06-21:
you seem to have commented out the argument nnights=5,
GGIR then falls back on the default which is 7 nights, this may confuse the code because there are not 7 nights in any of your files.

You do not have a waking up time for the last night. I think this could be problematic, because if I remember correctly GGIR requires at least a column to be present for both going to bed and waking up times. You could start by adding that column, if this does not work then remove column J

DiegoGBassani, 2017-06-22:
What I did was delete the bed time for the last night, and set the nnights argument to =4, but it still gives the same error (Part 4 Error in `[.data.frame`(S, , coln1 + ((i - 1) * 2) + 1) : undefined columns selected).

vincentvanhees, 2017-06-22:
Can you post here (or email me) example data from one participant for whom you have diary data records for all nights? For example person 4. I need the milestone files (.RData-format) for part 1, 2 and 3.

DiegoGBassani, 2017-06-22:
So what is happening is that I do not get the error anymore - but the person who is running the same code using the server in Pelotas, gets the error. We are very very puzzled with it…
I will e-mail you the milestone data.
D

closed 2017-07-22 by vincentvanhees