wadpac/GGIR #22: part5.R time_num at line 656

JianWang2016, 2017-06-07:
Vincent:
    An error encountered with Part 5.  It is the line at 656-658 of part5.R.   Any insight where could cause this?  Your help is greatly appreciated.

Part 2
 1
________________________________________________________________________________________________________________________________________________________________
Part 3
 1
________________________________________________________________________________________________________________________________________________________________
Part 4
 1
________________________________________________________________________________________________________________________________________________________________
Part 5
 1
Error in if (time_num < 12) time_num = time_num + 24 : 
  missing value where TRUE/FALSE needed
>

JianWang2016, 2018-03-13:
Dear all,  Just want to follow up on this issue.  It continues in the release of 1.5.17.  Any insight is greatly appreciated.

vincentvanhees, 2018-03-17:
I may have fixed this now, and will include this in the next release version of GGIR on CRAN in the upcoming days. Let me know if the problem persists.

closed 2018-03-17 by vincentvanhees