wadpac/GGIR #25: plot to check data quality only output 7 days

JianWang2016, 2017-06-07:
Hi Vincent,

the QC file named: "plots to check data quality 1.pdf" is very useful to inspect non wear in the data.  However, it only has 7 days of data while I do see 14 days of data from other reports.   Any advice is greatly appreciated.

-Jian

PedroMVidal, 2017-08-25:
Hi VIncent, Hi Jian

We have a similar issue, but regarding the output by do.report=c(5). The output shows data for the first 12 days, not the whole 14 days. We increased the part 2 parameters to 
maxdur=15
and the part 4 parameters to  nnights = 14,

But the problem persists
Thanks a lot

Pedro

vincentvanhees, 2019-01-10:
Hi both,
I am not able to reproduce these issues now, so I think they have been addressed since you posts in 2017. Definitely a lot of progress has been made in relation to getting the right number of days out of GGIR. If you still encounter these problems then please re-open this issue.
Best wishes, Vincent

closed 2019-01-10 by vincentvanhees