wadpac/GGIR #27: Incomplete output files

Lhurter, 2017-06-15:
Hi Vincent, 

I helped my colleague run her ActiGraph files through GGIR parts 1-4. However, we don't get the right output files. In the "results" folder there is only a "QC" folder but no summary files. This is the error message at the end: Error in matrix(" ", 1, (ncol(SUMMARY) - ncol(SUM$summary))) : invalid 'ncol' value (< 0). 

Can you shine any light on where we might have gone wrong?

Kind regards, 
Liezel

vincentvanhees, 2017-06-20:
Hi Liezel, I just replied to this question in the google group, but looks like you are providing more info here.
Please find my answer here: https://groups.google.com/d/topic/rpackageggir/wnU_LjZooI8/discussion
Vincent

Lhurter, 2017-06-22:
Thanks Vincent, it now works perfectly!
Liezel