wadpac/GGIR #30: Problems running Part2

DeboarahAnnMcCann, 2017-06-28:
Hi Vincent
I hope you can help? I can run Part1 fine, however when I run Part1 and 2 together, I receive this error message:
[image]

Please can you offer any advice?

Many thanks 
Debbie

vincentvanhees, 2017-07-02:
Does folder 'C:/GGIR/output_2g_control' exists?
Is there a file inside the folder 'C:/GGIR/output_2g_control/meta/basic'?

Either no data is saved after part 1 or the data is save somewhere else.
Some suggestions:
- Remove the _ from the data folder name. It might be that this confuses the code.
- Check that you have read&write permissions to the folders or drive you are saving to
- Try using a different folder location

Please let me know whether any of these suggestions addresses the issue

DeboarahAnnMcCann, 2017-07-04:
Hi Vincent

Thank you for your reply. In response to your queries:

Yes, part 1 of GGIR seems to work fine, therefore producing a R meta file for each raw file. A results folder is also produced but inside only has a QC folder inside. In the results folder is a file containing the filelocationkey and R sessioninfo_part1 file but no part2_daysummary or part_2summary.

I have removed all _from raw and output folders, still part 2 failed with the same error message.
I have checked the read & write permissions, still part 2 failed with the same error message.
I have changed the location and drive of the folder location and still part 2 failed with the same error message.

Any further help that you could offer is greatly appreciated.

Kind Regards
Debbie

vincentvanhees, 2017-07-05:
Can you upload the .RData file that comes out of part1 here? or if it includes patient sensitive information then email it to me.

jmkurka83, 2017-07-05:
Hi Vincent,
For ease, we can use this thread for communicating. I've emailed my sessioninfo_part1 data if you'd like a 2nd example to Debbie's post. Github won't allow a .RData upload. Let me know if you'd like it uploaded in another way.
Thank you!
Jonathan

DeboarahAnnMcCann, 2017-07-05:
Hi Vincent, Jonathan

I have sent 10 RData files via Dropbox. I am currenty running batches of
20-30 and have 8 batches of GENEActiv data and 8 of ActiGraph data, all
have given the same error code for Part2, therefore any help you can offer
is greatly appreciated.

Many thanks
Debbie

vincentvanhees, 2017-07-07:
Please see my response to the same issue here: https://github.com/wadpac/GGIR/issues/32. In short, the outputdir argument is probably not correctly specified.

vincentvanhees, 2017-07-22:
I assume this issue can be closed with listed in solution #32 and/or googlegroup:
https://groups.google.com/forum/#!topic/rpackageggir/RY8wzp7_Dos/discussion.
Thanks

closed 2017-07-22 by vincentvanhees