wadpac/GGIR #32: Part 2 Error : "error in readChar"

jmkurka83, 2017-07-04:
Hello,
I've come across an error that I haven't been able to solve. I've been running GGIR with success for some time, however within the past week it will not run past part 2 without throwing this error:
Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
In addition: Warning message:
In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file 'myfolderpath/output_csvfiles_test/meta/basic/NA', probable reason 'No such file or directory'

I've uninstalled R and R studio and updated all GGIR packages. The output and directory folder paths are accurate. Does anyone have suggestions?
Thank you!
- Jonathan

DeboarahAnnMcCann, 2017-07-04:
This is the same error as I am encountering-hopefully it can be resolved!
Debbie

vincentvanhees, 2017-07-05:
Thanks for reporting this. I have not changed GGIR recently, so I am not sure what is causing it. I will have a look at it and get back to you.

vincentvanhees, 2017-07-07:
Thanks for sending the example files Jonathan. I have been able to process them, and I was able to reproduce your error message by incorrectly specifying the outputdir directory. Please review that outputdir is specified correctly.

jmkurka83, 2017-07-07:
Hi Vincent,
I've checked and my outputdir specification matches the folder that eventually houses the output results. I'm still getting the same error. Does having other items within the outputdir folder have an impact? Does the working directory have an impact as well. Any assistance is appreciated!
Thank you
Jonathan

jmkurka83, 2017-07-10:
I've actually noticed that the error I'm not getting is not exactly the same. The error reads the exact same as posted above, except that instead of 'No such file or directory' it states 'Invalid argument.
Jonathan

DeboarahAnnMcCann, 2017-07-11:
Hi Vincent, thank you for your suggestion. Unfortunately this has not
resolved the issue.
The outputdir directory has been correctly specified, as Part1 is all
produced and documents are within the file. Unless a separate file needs to
be specified for Part2? However when I have ran this in the past this
hasn't been the case. My error message remains the same, any help would be
appreciated.

Kind Regards
Debbie

Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection

In addition: Warning message:

In readChar(con, 5L, useBytes = TRUE) :

  cannot open compressed file 'E:/GGIR/output_raw3amobile/meta/basic/NA',
probable reason 'No such file or directory'

vincentvanhees, 2017-07-11:
Could have to do with your file permission settings. Are you the admin? If you are not, ask the admin to give you full permissions for the directory where your data sits.

jmkurka83, 2017-07-11:
I'm the admin for both of the computers I've tried to run GGIR on. None of my admin settings have changed since it last worked about 3-4 weeks ago.

vincentvanhees, 2017-07-13:
Sorry, i really cannot help you further on this. My time is limited and it sounds like a standard IT issue and not a bug in GGIR.

DeboarahAnnMcCann, 2017-07-14:
Hi Vincent
My issue has now been resolved, I was just waiting on my final files
processing before I got in touch.
Thank you for your help with this matter
Debbie

DeboarahAnnMcCann, 2017-07-14:
Hi Jonathan

My issue has now been resolved, if you are still struggling please let me
know and Ill bullet point the things I changed as this may help. As Vincent
suggested this seems to be an IT error.

Kind Regards
Debbie

JianWang2016, 2017-07-14:
Hi Debbie:

If you don't mind, could you please share what changes you made to resolve the issue?  I think it would be helpful for others.

Thanks much.

DeboarahAnnMcCann, 2017-07-14:
Hi not at all. I tried many things that I had changed in the past and been
unsuccessful along with a few extra, I do hope that this can help others:

-Removed all software (only used R 64 bit previously)

-re installed all software (installed both 32 and 64 bit)

-Deleted the R script

-downloaded and saved new R script

-changed all files (not just the file name) by doing the following:
  -create new raw file
  -copied and pasted contents only from old raw file (this can take a while)
  -create new output folder
  -copy and paste old output_raw folder (from inside old output folder)
  - rename output_folder to reflect new raw folder (output_newrawfolder

When using the software I then loaded all dependencies (instead of relying
on GGIR to load them)
bitops, data.table, GENEread, GGIR, MASS, Matlab, mmap, Rcpp, signal, zoo

I'm not saying this is the best way to do things but it worked for me. I
hope this helps.

Kind Regards

closed 2017-07-22 by vincentvanhees