wadpac/GGIR #44: generating csv epoch files for short lab data files

Alex-Row, 2017-07-25:
Hi Vincent,

I'm having trouble getting the epoch2csv files from short (approx 1 h) lab GENEActiv and ActiGraph files, it is ok for longer files - approx 2 h. A while back I previously successfully obtained these for a couple of the exact same files (version 1.5-3), but have been unable to replicate.

Main points:
I'm using GGIR 1.5-10 and have tried 1.5-9 and 1.4 and had the same result with each (I have been unable to install 1.5-3 for some reason).
I'm entering backup calibration coefficients, which are being used
I've set window sizes to (1,600,600), (1,300,300), (5,600,600) and (5,300,300) - same result with all
do.imp = FALSE
I've tried chunk sizes of 0.2 and 1.0

With the GENEActiv files the meta basic files are created, but for files shorter than 2 h (<1 MB) they are empty - only about 2 KB rather than about 200 KB
The epoch csv files are therefore only created for the files longer than 2 h
The warning is: Warning (2): data in block too short for doing non-wear detection

With the ActiGraph files, the meta basic file is created for files over about 13 MB, but as soon as it hits a shorter file it stops and says:
stop reading because there is not enough data in this block
Error in names(spheredata) = c("Euclidean Norm", "meanx", "meany", "meanz",  : 
  'names' attribute [8] must be the same length as the vector [7]

Is it still possible to create csv epoch files for files of about an hour? And is it possible to 'turn off' the non-wear detection for short files?

Any advice much appreciated.

Cheers,
Alex

vincentvanhees, 2017-07-27:
Hi Alex, there is indeed a minimum measurement duration hard-coded in GGIR. I only added it with the assumption that nobody would ever be interested in less than 2 hours of free-living data. I will look into this. Thanks, Vincent

Alex-Row, 2017-07-27:
Thanks, Vincent.

Alex-Row, 2017-07-27:
Update. I've managed to solve this using GGIR 1.4 re-installed. I neglected to re-start R when I re-installed the earlier packages previously.

sjf69, 2017-08-01:
Hi Vincent, just to say I've got exactly the same situation as Alex (and used the same fix by installing v1.4). I was wondering if this will be resolved in future GGIR versions, to avoid users having to go back to v1.4 each time? Thanks, Stuart

vincentvanhees, 2017-08-06:
I have found the source of the problem and fixed it here in github. Our plan is to upload a new GGIR version 1.5-11 in the upcoming days.

closed 2017-08-06 by vincentvanhees