wadpac/GGIR #50: Version 1.5-12 does not provide the report part 5

PedroMVidal, 2017-09-05:
HI

I am using version 1.5-12 of GGIR and everything goes smoothly till the part 5 report is to be produced. Irrespective of the *.bin files used, I constantly get the same error message (screen copies attached). Should I use version 11 instead?
[Screen copies.docx](https://github.com/wadpac/GGIR/files/1276519/Screen.copies.docx)

vincentvanhees, 2017-10-02:
Hi Pedro, the error message reads to me as if part5 report generation struggles to combine the output from different measurements. Can you delete all RData files in the ms5.out folder and generate them again? Do you keep having the problem?
Always reprocess all files if you change your analyses configuration, otherwise you end up with milestone data that is half derived in one way and half in another way, this complicates generating the reports.

vincentvanhees, 2017-12-23:
Did this answer address your issues?