wadpac/GGIR #53: Error in readChar : cannot open the connection

jmkurka83, 2017-10-05:
Hello Vincent,
I've been troubleshooting this error since July after the most recent updates went into effect. I've been assured I have administrator rights and have added R to the antivirus exception list (both recommended).
This is the primary additional message, stopping GGIR leaving a meta data file in the 'basic' folder, but ms2.out is empty.

In addition: Warning message:
In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file 'C:/R/win-library/3.4/csvfiles_test/output_csvfiles_test/meta/basic/NA', probable reason 'No such file or directory'

After noticing that the 'ms2.out' folder was empty, I copied the meta file in 'basic' and pasted it into 'ms2.out'. The error remained the same, but the additional message now reads:

In addition: Warning messages:
1: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
2: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
3: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
4: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
5: In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file 'C:/R/win-library/3.4/csvfiles_test/output_csvfiles_test/meta/basic/NA', probable reason 'No such file or directory'

Could the issue be with my file names? Again, everything worked until the GGIR updates in the summer. Here is an example of my file names: 2051_(2013-04-11).csv
I've working with our IT department on solutions. Any insight would be much appreciated!
Thank you!
- Jonathan

jmkurka83, 2017-10-06:
I ran traceback() to try to pinpoint where the error occurs. It pointed to line 66 of part2 code.

vincentvanhees, 2017-10-08:
Hi Jonathan, thanks for reporting this. I am at a conference at the moment. 
One possible explanation could be that the new version of GGIR somehow is not able to handle part1 output generated with older versions of GGIR. I will try to look into this in the upcoming weeks.

jmkurka83, 2017-10-12:
Thank you, Vincent.
To update, I simplified the file names and still received the same error. For example, I changed the above file name to 2051_20130411 and set idloc = 2. Looking forward to hearing from you! Thank you again for all your work on this!

jmkurka83, 2017-10-19:
Hi Vincent. I tried running only part1 and received this message
Warning: First run g.shell.GGIR with mode = 4 to generate required milestone data
Error in g.shell.GGIR(mode = mode, datadir = datadir, outputdir = outputdir,
I wasn't sure if that was relevant so wanted to include this information. I've tried reverting back to version GGIR 1.4 with R version 3.4.0, but have trouble with the dependencies, or have come across the same error.

jmkurka83, 2017-11-10:
Hi Vincent,
I think we've located the issue. The issue seems to arise when trying to run only subsets of files using the f0 and f1 specifications. Line 18 of part2, fnames is populated with the list of files from that variable path derived from the files indicated in part1. If I understand this correctly, line 67 of part2 `load(paste(path,fnames[i],sep="")) #reading RData-file` results in an error if f0 is larger than fnames. My example is setting f0=651 and f1=800 (we have about 900 files to analyze). In this case, fnames totals 150, so part2 cannot run when looking for the file in position 651 (fnames[651]).

A potential fix is to change line 51 in part2 to initialize i from 1:length(fnames).

Our IT department head here at ASU College of Health Solutions worked with me on this and ultimately came up with this solution. Please let me know if there is anything I can do to help fix this error.
Thanks!
- Jonathan Kurka

vincentvanhees, 2018-03-17:
Thanks Jonathan, I seem to have forgotten about this message! I have now incorporated a slightly different approach because your solution makes f0 and f1 redundant for part2, which is not ideal. In the new solution f0 and f1 are corrected based on the length of fnames. Thanks again for looking into this.

closed 2018-03-17 by vincentvanhees

prinsherbert, 2019-07-31:
I seem to get the same error on missing files in the **output** folder, is this a bug or am I using the library wrong? I would expect the outputfolder is supposed to be empty. Moreover, creating respectively touching the missing folders and files does not solve the issue.

    $ cat test_ggir.R 

    require(GGIR)

    g.shell.GGIR(mode=c(2), datadir='/media/veracrypt2/PhysicalActivity/gt3X FILES/', outputdir='/home/herbert/RuG/actigraph/ggir/')

    $ ls '/media/veracrypt2/PhysicalActivity/gt3X FILES/' | head -n 5
    CSPF 001 (2017-11-13) - Intake60sec.agd
    CSPF 001 (2017-11-13) - Intake.gt3x
    CSPF 001 (2017-11-23) - Intake2.gt3x
    CSPF 001 (2017-11-23) - Intake60sec2.agd
    CSPF 006 (2018-12-03) - Intake60sec.agd
    $ ls '/home/herbert/RuG/actigraph/ggir/'
    $ Rscript test_ggir.R 
    Loading required package: GGIR

       Help sustain GGIR into the future 
       Check out: https://www.movementdata.nl/how-to-help-sustain-ggir 
       for more information. 

    ________________________________________________________________________________
    Part 2
     1Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
    Calls: g.shell.GGIR -> g.part2 -> load -> readChar
    In addition: Warning messages:
    1: In dir.create(file.path(metadatadir, ms2.out)) :
      cannot create dir '/home/herbert/RuG/actigraph/ggir//output_gt3X FILES//meta/ms2.out', reason 'No such file or directory'
    2: In readChar(con, 5L, useBytes = TRUE) :
      cannot open compressed file '/home/herbert/RuG/actigraph/ggir//output_gt3X FILES/meta/basic/NA', probable reason 'No such file or directory'
    Execution halted
    $

vincentvanhees, 2019-07-31:
GGIR cannot handle .gt3x or .agd file, you will need ActiLife software to convert the data to csv format, which GGIR should then be able to read.

Further, you will need to use mode = 1, because GGIR is a process of 5 parts where each part relies on the output of the preceeding part.

prinsherbert, 2019-08-05:
> GGIR cannot handle .gt3x or .agd file, you will need ActiLife software to convert the data to csv format, which GGIR should then be able to read.

I tried both the CSV-data and the gt3x / agd folder, and both failed. In the meantime I found the documentation mentioning CSV for Actigraph, Thanks for pointing this out.

> Further, you will need to use mode = 1, because GGIR is a process of 5 parts where each part relies on the output of the preceding part.

Clear, that makes sense.

nkuzik, 2019-11-08:
I am receiving the same error message when attempting to run the basic shell function on a folder with ActiGraph csv files (tested both timestamped and non-timestamped):

```
g.shell.GGIR(datadir="C:/Users/nkuzi/Downloads/Test", 
outputdir="C:/Users/nkuzi/Downloads/Results")

   Please refer to GGIR by reporting the version number and citing
   Migueles et al. 2019 J Meas Phys Beh. See also: 
   https://cran.r-project.org/package=GGIR/vignettes/GGIR.html#citing-ggir 

________________________________________________________________________________
Part 1

Checking that user has read access permission for all files in data directory: Yes
Checking that user has write access permission for directory specified by argument outputdir: Yes

 Busy processing ... see C:/Users/nkuzi/Downloads/ProxDev Raw/Results/output_T/meta/basic for progress

Errors and warnings for 1.csv$message
[1] "'from' must be a finite number"

$call
seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) * 
    ws3)), by = ws3)

________________________________________________________________________________
Part 2

 Busy processing ... see C:/Users/nkuzi/Downloads/ProxDev Raw/Results/output_T/ms2 for progress

Errors and warnings for NA$message
[1] "cannot open the connection"

$call
readChar(con, 5L, useBytes = TRUE)

Errors and warnings for NA$message
[1] "cannot open the connection"

$call
readChar(con, 5L, useBytes = TRUE)

________________________________________________________________________________
Part 3

 Busy processing ... see C:/Users/nkuzi/Downloads/ProxDev Raw/Results/output_T/ms3 for progress

Errors and warnings for NA$message
[1] "cannot open the connection"

$call
readChar(con, 5L, useBytes = TRUE)

Errors and warnings for NA$message
[1] "non-character argument"

$call
strsplit(fname, "eta_")

________________________________________________________________________________
Part 4
 1Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
In addition: Warning message:
In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file 'C:/Users/nkuzi/Downloads/ProxDev Raw/Results/output_T/meta/ms3.out/NA', probable reason 'No such file or directory'
```
I have tried a number of things with no success, including:
- Adding ression.exe, rstudio.exe, Rterm.exe as exceptions in Windows Defender Antivirus scans
- Making sure that GGIR was downloaded with all dependencies
- Renaming my test files sequentially (e.g., 5c_proxdev.csv, 17c_proxdev.csv; renamed to 1.csv, 2.csv)
- Running RStudio as Admin

Some more possibly useful info:
```
> sessionInfo()
R version 3.6.1 (2019-07-05)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 17763)

Matrix products: default

locale:
[1] LC_COLLATE=English_Canada.1252  LC_CTYPE=English_Canada.1252   
[3] LC_MONETARY=English_Canada.1252 LC_NUMERIC=C                   
[5] LC_TIME=English_Canada.1252    

attached base packages:
 [1] parallel  tools     compiler  stats     graphics  grDevices utils    
 [8] datasets  methods   base     

other attached packages:
[1] doParallel_1.0.15 iterators_1.0.12  foreach_1.4.7     codetools_0.2-16 
[5] data.table_1.12.6 Rcpp_1.0.2        GGIR_1.10-7  
```
Thanks in advance!

HelenEke, 2019-11-18:
Hi Nkuzik,
I had the same problem using Actigraph-files. In my case, GGIR didn't recognize the date/time in the csv-file. This is solved with an update of GGIR!

install.packages("remotes")
remotes::install_github("wadpac/GGIR")

Regards,
Helén

nkuzik, 2019-11-20:
Hi Helén,

Thanks for this, your solution worked! Perhaps my issue was downloading via:

```install.packages("GGIR",dependencies=TRUE)```

Or perhaps it was because your suggestion forced me to download RTools? Either way, it is working now.

Thanks again!!

vincentvanhees, 2019-11-21:
Great to see that you figured it out between each other:

`install.packages("GGIR",dependencies=TRUE)`

installs GGIR from CRAN, while

```
install.packages("remotes")
remotes::install_github("wadpac/GGIR")
``` 
or 
```
install.packages("devtools")
remotes::install_github("wadpac/GGIR")
```

installs GGIR from GitHub.

On a side note: I am aware that it is time that I upload these recent bug fixes to CRAN. Will try to find some time for that in the upcoming week.

commit e032ab33 (2018-03-17): Added a check to f0 and f1 valuesinside g.part2, closes #53
commit b571f8fc (2020-06-18): Added a check to f0 and f1 valuesinside g.part2, closes #53