wadpac/GGIR #54: GT3X-csv data, g.part1(): Error in g.getstarttime(datafile = datafile, P = P, header = header, mon = mon,  :    object 'starttime0' not found

yaojiali, 2017-11-01:
Hi Vincent,

I am trying to use GGIR to analysis the csv file converted from .gt3x file (timestamps not exported). However, when I use the g.part1 function, I got the following error:
------
Loading block: 1Error in g.getstarttime(datafile = datafile, P = P, header = header, mon = mon,  : 
  object 'starttime0' not found
-----
If possible, could you kindly let me advise me on this?  Thanks a lot!!!

Regards,
Jiali 

==============the whole output chunk:
> g.part1("./in", "./out", f0 = 1, f1= 1, desiredtz = "Asia/Shanghai")
[1] "P1 file1"
[1] "---------------------------------------------"
[1] "investigate calibration of the sensors..."

Loading block: 1 2 3 4 5 6 7
----------------------------------------
Summary of autocalibration procedure:

Status:
[1] "recalibration done, no problems detected"

Calibration error (g) before:
[1] 0.01400207

Callibration error (g) after:
[1] 0.005550402

Offset correction:
[1] -0.007008119 -0.006828931 -0.010021307

Scale correction:
[1] 0.9972895 1.0184210 0.9945531

Number of hours used:
[1] 84

Number of 10 second windows around the sphere:
[1] 24203

Temperature used (if available):
[1] FALSE

Temperature offset (if temperature is available):
[1] 0 0 0

----------------------------------------
[1] "get meta data..."

Loading block: 1Error in g.getstarttime(datafile = datafile, P = P, header = header, mon = mon,  : 
  object 'starttime0' not found

yaojiali, 2017-11-01:
Below is the header i got when I use g.inspectfile(): 
$header
                                                                                                                                                                                 value
First line                               X.............Data.File.Created.By.ActiGraph.GT3X..ActiLife.v6.11.7.Firmware.v1.9.0.date.format.dd.MMM.yy.at.80.Hz..Filter.Normal............
Serial Number:                                                                                                                                                           MOS2D30160316
Start Time                                                                                                                                                                    12:00:00
Start Date                                                                                                                                                                   18-Oct-16
Epoch Period (hh:mm:ss)                                                                                                                                                       00:00:00
Download Time                                                                                                                                                                 06:16:14
Download Date                                                                                                                                                                16-Nov-16
Current Memory Address:                                                                                                                                                              0
Current Battery Voltage: 4.18     Mode =                                                                                                                                            12

Thanks a lot!!

Regards,
Jiali

yaojiali, 2017-11-05:
I figured out why.... should be the date format "dd-mmm-yy" that cause the problem. I changed the date format in the raw file to "mm/dd/yyyy", then it works~~

closed 2017-11-05 by yaojiali

vincentvanhees, 2017-12-23:
Thank you Jiali for reporting this, I just created a new issue for this to be addressed in one of the upcoming releases of GGIR.

yaojiali, 2017-12-24:
Thanks you very much Vincent~~ Merry Christmas and wish you a happy new year~~

Regards,
Jiali

vincentvanhees, 2018-03-17:
This fix will also be included in the next release on CRAN in the upcoming days.

commit b14f4d49 (2018-03-17): Fixes #54
commit 829bebe6 (2018-03-17): Fixes #54 and Fixes #58 and fixes #75
commit 3ff83b8c (2020-06-18): Fixes #54 and Fixes #58 and fixes #75
commit 8a2bb42d (2020-06-18): Fixes #54