wadpac/GGIR #55: Processing errors

kellehek, 2017-11-30:
Hi,

I've been hitting a number of stumbling blocks that I'm hoping to get some help with. 

I have ~1600 pupils with 5 days wear split into 40 folders (one per school). Most have ran fine, but 2 schools keep failing. One in part 4 with the error message "Error in 1:nrow(spo) : argument of length 0
Calls: g.shell.GGIR -> g.part4" and the other in part 5  with the error message "Error in as.POSIXlt.character(w0[k], tz = desiredtz) : character string is not in a standard unambiguous format
Calls: g.shell.GGIR -> g.part5 -> as.POSIXlt -> as.POSIXlt.character" 

Additionally, on inspecting the outputs from the different parts for the ones that did run, I seem to have 5 days wear time in part 2 and part 4, but only 1-2 days wear time when they combine in part 5. Any suggestions as to where I might be going wrong would be gratefully received? 

Many thanks, 

Kiya

vincentvanhees, 2017-12-06:
At the moment I have limited time to respond to this and other issues. I am in the process of setting up a paid consultancy service, where I would ask you and others for a compensation of my time to help you with your issue(s), but also to improve GGIR to avoid similar issue(s) for future users. I hope to get back to you with more detailed information soon.
Best wishes, Vincent

vincentvanhees, 2017-12-23:
- Sounds like a timestamp or date format difference. Did part 1-4 ran fine and does the output from part 1-4 look plausible?
- Do you notice anything different in the formatting of the timestamps from those schools compared with the other schools?
- I am aware of the part5 issue. Part5 currently gives less days than part 4 because it defines days from wake-up till wake-up the next day, or from noon-noon. At the moment I implemented it to ignore any day for which there is not both a night before and after.  Possibly this is too conservative and needs to be looked at.

vincentvanhees, 2018-01-17:
Did my response help you address the issues?

kellehek, 2018-01-17:
Dear Vincent, Thank you so much for your response and sorry for my lack of reply. Due to the timescales of our study, we have had to go down a different route for processing the geneactivs. I'm afraid my processing knowledge was just not good enough to tackle the number of issues I was facing within the time I had to do so. Please do let me know if you set up paid consultancy as I may need to access it in the future if we use geneactivs again. Kind regards, Kiya

vincentvanhees, 2018-02-09:
Hi Kiya, yes I am now set up to provide paid consultancy: www.movementdata.nl
Best regards, Vincent

closed 2018-02-09 by vincentvanhees