wadpac/GGIR #57: Failure to read CWA from Axtivity UKBiobank Data

muschellij2, 2017-12-13:
The example `.cwa` file from the UKBiobank data fails to read in using `g.cwaread`.  The file is loacted at https://biobank.ctsu.ox.ac.uk/crystal/refer.cgi?id=131620, with a hard link to the zip file at https://biobank.ctsu.ox.ac.uk/crystal/examples/accsamp.zip.  

After unzipping the file and running:
```r
library(GGIR)
x = g.cwaread("accsamp.cwa", progressBar=TRUE, end = Inf)
```
the following error is 
```
  |====================================================================  |  97%[1] "last block of data"
Error in accelRes[pos:last, ] = tmp : replacement has length zero
```

I set `end = Inf`, because in the code in `g.cwaread` (https://github.com/wadpac/GGIR/blob/master/R/g.cwaread.R#L274) has:
```
    if (is.numeric(end)) {
        end = end * pageLength
        if (end > numDBlocks * 120) {
            end = numDBlocks * 120
        }
        end = origin + end * step
    }
```

This has failed on data from the UKBiobank as well: `1234520_90001_0_0.cwa` and `2942234_90001_0_0.cwa`, for example.  I can't seem to pinpoint exactly the issue.  It occurs in the last block of data, when the `resample` command is called, but can't seem to figure out a fix for a pull request.

The problem persists with the CRAN version on the current github commit 832d1848dd0f7e890c0f069145c80b674c58a855

Referencing @Mirkes as the author of `g.cwaread`.

Mirkes, 2017-12-14:
Dear John,

I'll analyse this problem ASAP. Unfortunately this week is completely busy.

Best regards,
Evgeny

Mirkes, 2017-12-18:
Dear John,

Many thanks for your message. Now problem with reading of files is fixed. New version of g.cwaread is uploaded to github.

Best regards,
Evgeny

closed 2017-12-18 by Mirkes

vincentvanhees, 2017-12-21:
thanks Evgeny and John, I aim to get a new release of GGIR uploaded to CRAN in the upcoming days

muschellij2, 2020-02-18:
This worked for a bit, but now `tz` issues:

``` r
zipfile = tempfile(fileext=".zip")
download.file("http://biobank.ndph.ox.ac.uk/showcase/showcase/examples/accsamp.zip", 
              destfile=zipfile)
fname = unzip(zipfile, files = "accsamp.cwa", exdir = tempdir())

library(GGIR)
# x = g.cwaread("accsamp.cwa", progressBar=TRUE, end = Inf)
x = g.cwaread(fname, progressBar=TRUE, end = Inf)
#> Error in strptime(xx, f, tz = tz): invalid 'tz' value
```

<sup>Created on 2020-02-18 by the [reprex package](https://reprex.tidyverse.org) (v0.3.0.9001)</sup>

<details style="margin-bottom:10px;">

<summary>Session info</summary>

``` r
sessioninfo::session_info()
#> ─ Session info ───────────────────────────────────────────────────────────────
#>  setting  value                       
#>  version  R version 3.6.2 (2019-12-12)
#>  os       macOS Mojave 10.14.6        
#>  system   x86_64, darwin15.6.0        
#>  ui       X11                         
#>  language (EN)                        
#>  collate  en_US.UTF-8                 
#>  ctype    en_US.UTF-8                 
#>  tz       America/New_York            
#>  date     2020-02-18                  
#> 
#> ─ Packages ───────────────────────────────────────────────────────────────────
#>  package     * version     date       lib source                             
#>  assertthat    0.2.1       2019-03-21 [1] CRAN (R 3.6.0)                     
#>  backports     1.1.5       2019-10-02 [1] CRAN (R 3.6.0)                     
#>  cli           2.0.1       2020-01-08 [1] CRAN (R 3.6.0)                     
#>  crayon        1.3.4       2017-09-16 [1] CRAN (R 3.6.0)                     
#>  data.table    1.12.8      2019-12-09 [1] CRAN (R 3.6.0)                     
#>  digest        0.6.23      2019-11-23 [1] CRAN (R 3.6.0)                     
#>  evaluate      0.14        2019-05-28 [1] CRAN (R 3.6.0)                     
#>  fansi         0.4.1       2020-01-08 [1] CRAN (R 3.6.0)                     
#>  fs            1.3.1       2019-05-06 [1] CRAN (R 3.6.0)                     
#>  GGIR        * 1.10-4      2019-09-09 [1] CRAN (R 3.6.0)                     
#>  glue          1.3.1       2019-03-12 [1] CRAN (R 3.6.0)                     
#>  highr         0.8         2019-03-20 [1] CRAN (R 3.6.0)                     
#>  htmltools     0.4.0       2019-10-04 [1] CRAN (R 3.6.0)                     
#>  knitr         1.26.1      2020-01-05 [1] Github (muschellij2/knitr@f5af631) 
#>  magrittr      1.5         2014-11-22 [1] CRAN (R 3.6.0)                     
#>  pillar        1.4.3       2019-12-20 [1] CRAN (R 3.6.0)                     
#>  pkgconfig     2.0.3       2019-09-22 [1] CRAN (R 3.6.0)                     
#>  purrr         0.3.3       2019-10-18 [1] CRAN (R 3.6.0)                     
#>  Rcpp          1.0.3       2019-11-08 [1] CRAN (R 3.6.2)                     
#>  reprex        0.3.0.9001  2020-01-05 [1] Github (tidyverse/reprex@5ae0b29)  
#>  rlang         0.4.2       2019-11-23 [1] CRAN (R 3.6.0)                     
#>  rmarkdown     2.0.7       2020-01-17 [1] Github (rstudio/rmarkdown@2faf16a) 
#>  rstudioapi    0.10.0-9003 2020-01-05 [1] Github (rstudio/rstudioapi@abe596d)
#>  sessioninfo   1.1.1       2018-11-05 [1] CRAN (R 3.6.0)                     
#>  stringi       1.4.5       2020-01-11 [1] CRAN (R 3.6.0)                     
#>  stringr       1.4.0       2019-02-10 [1] CRAN (R 3.6.0)                     
#>  styler        1.1.1       2019-05-06 [1] CRAN (R 3.6.0)                     
#>  tibble        2.1.3       2019-06-06 [1] CRAN (R 3.6.0)                     
#>  withr         2.1.2       2018-03-15 [1] CRAN (R 3.6.0)                     
#>  xfun          0.11        2019-11-12 [1] CRAN (R 3.6.0)                     
#>  yaml          2.2.0       2018-07-25 [1] CRAN (R 3.6.0)                     
#> 
#> [1] /Library/Frameworks/R.framework/Versions/3.6/Resources/library
```

</details>

vincentvanhees, 2020-02-18:
Hi John,
You should use "desiredtz" argument to specify timezone if the accelerometers were set up in the same timezone as in which the recording took place.

The configtz is new since last year and it currently only works for AX3 cwa data.
It allows users to configure their accelerometers in one timezone, send them to a study participant in another timezone. I have tested this with a client who had to deal with such situation, so it should work.

desiredtz = timezone of data collection
configtz = timezone were accelerometer was configured

vincentvanhees, 2020-02-18:
Note that ` if (length(configtz) == 0) configtz = desiredtz` at the top of g.cwaread handles `configtz = c()`

muschellij2, 2020-02-18:
I see the code, but default should likely be desiredtz = "", so that it
doesn't fail with as.POSIXlt and is the same default in as.POSIXlt, which
is tz = ""
Best,
John

vincentvanhees, 2020-02-25:
thanks John, I agree. Would be good to get that checked and fixed. I have opened a new issue for it.