wadpac/GGIR #59: error of "g.calibrate -> scale -> scale.default -> sweep Execution halted"

yaojiali, 2017-12-24:
Hi Vincent,
I am running GGIR on csv files from Actigraph, for some data files, I have the following error: 

Error in FUN(x, aperm(array(STATS, dims[perm]), order(perm)), ...) : 
"non-numeric argument to binary operator
Calls: g.shell.GGIR ... g.part1 -> g.calibrate -> scale -> scale.default -> sweep
Execution halted".

If you have ideas why this would happen to some csv files, could you kindly advise?

Thanks a lot!

Regards,
Jiali

vincentvanhees, 2018-01-17:
Hi Jiali, I have not seen this error before. It may be related to unexpected data format. Do you observe any differences in the formatting of the files for which the analysis runs successfully and those for which it does not?

yaojiali, 2018-01-30:
Hi Vincent,

Thank you so much for our reply! I processed 140 datasets collected from
Actigraph using GGIR in HPC. Recently my colleague used ActiLife to process
these datasets. Based on the outputs from ActiLife,  6 out of the 9 files
that I have this error message during calibration process are likely to
have short wear time, thus a large part of these dataset collected are 0 in
the x, y, z axis. I tried to look back to the source code of g.calibrate,
but I can't find where it might go wrong... I am thinking about trimming
the raw csv files (removing the last chunk of non-wear data at the end),
then run the script again. Yet I am not sure if I should change the times
in the csv header. if possible, could you kindly advise me on this? Thank
you so much!

Have a great week!

Best,
Jiali

yaojiali, 2018-02-26:
This problem is solved after I cut the raw csv file and ajust the date/time in the file header, only keep the data from one day before the participant-reported wear date to one day after the participant reported end date. 

hope this is useful~~~

Minjingy, 2022-01-20:
> Hi Vincent, Thank you so much for our reply! I processed 140 datasets collected from Actigraph using GGIR in HPC. Recently my colleague used ActiLife to process these datasets. Based on the outputs [...]

Hi yaojia, I am also dealing with accelerametor data using GGIR in HPC, but I received an error "Error in makePSOCKcluster(names = spec, ...) : Cluster setup failed. 38 of 39 workers failed to connect." Have you ever met this problem in HPC? if not, could you tell me what kind of R version did you use to process these datasets?