wadpac/GGIR #60: error when runing g.shell.GGIR with "mode=c(1), visualreport = TRUE, ..."

yaojiali, 2017-12-24:
Hi Vincent,

I was running GGIR on csv files from Actigraph, and I ran with g.shell.GGIR(). I first filled in all the arguments for g.shell.GGIR function that intends for mode = c(1, 2, 3, 4, 5). to make sure that each part in run after the previous one has finished, I change the mode = c(1) when I ran it in HPC cluster. Yet I kept getting message of error after g.getmeta. Then I looked at the source code for function g.shell.GGIR(), I found that the function doesn't seem to allow for "mode=c(1), visualreport = TRUE, ...", this could be caused by the codes between line 240 and 247.

[image]

if possible, could you kindly have a look at it~~
Thanks a lot!

Regards,
Jiali

vincentvanhees, 2018-01-17:
Hi Jiali,
This is because you are probably using do.report=c(4) in your call of g.shell.GGIR, which then complains that mode =4 was not done yet. If you turn this of or replace it by do.report = c(), it should fix your problem.
Best, Vincent

yaojiali, 2018-01-29:
Thank you very much Vincent!

closed 2018-02-26 by yaojiali