wadpac/GGIR #61: GGIR package won't install

TanjaHarrison, 2018-01-12:
Good afternoon,

I have downloaded the GGIR_1.5-12.tar.gz package, but for some reason it will not install. I have tried two different laptops for this. Every time I try I get the following message below. 
How could I solve this, please?
Many thanks,
Tanja

> install.packages("C:/GGIR_1.5-12.tar.gz", repos = NULL, type = "source")
* installing *source* package 'GGIR' ...
** package 'GGIR' successfully unpacked and MD5 sums checked
** libs

*** arch - i386
Warning: running command 'make -f "C:/PROGRA~1/R/R-32~1.2/etc/i386/Makeconf" -f "C:/PROGRA~1/R/R-32~1.2/share/make/winshlib.mk" SHLIB_LDFLAGS='$(SHLIB_CXXLDFLAGS)' SHLIB_LD='$(SHLIB_CXXLD)' SHLIB="GGIR.dll" OBJECTS="RcppExports.o init.o numUnpack.o resample.o"' had status 127
ERROR: compilation failed for package 'GGIR'
* removing 'C:/Program Files/R/R-3.2.2/library/GGIR'
Warning in install.packages :
  running command '"C:/PROGRA~1/R/R-32~1.2/bin/x64/R" CMD INSTALL -l "C:\Program Files\R\R-3.2.2\library" "C:/GGIR_1.5-12.tar.gz"' had status 1
Warning in install.packages :
  installation of package ‘C:/GGIR_1.5-12.tar.gz’ had non-zero exit status

vincentvanhees, 2018-01-17:
R packages come in different formats depending on your operating system.  In the github releases I usually only include the source file build under Linux, you can clone the reposity and build your own installation file on other platforms.

A possibly easier route for you as user is to download the latest release from CRAN directly: install.packages(GGIR, dependencies = TRUE), this also makes it easier to keep the package up to date..

Does this answer your question? If not, can you clarify where you downloaded the tar.gz and in what operating system you are trying to install it?

TanjaHarrison, 2018-01-21:
Hi,

Many thanks for your advice.

Unfortunately, Install.packages  has not worked either.

I have tried two different computers:

1)      OS Windows 7; tried to install to C drive

2)      OS Windows 10 Enterprise, also C drive

I have since tried to run my script on a computer that already had GGIR installed but have run into problems here, too.

I will open a new tread for this.

Many thanks,
Tanja

[cid:image001.gif@01D20DB2.997272C0]<http://www.ljmu.ac.uk/>

Tanja Harrison BSc. (Hons) ANutr AFHEA SEDA
PhD candidate and Registered Associate Nutritionist
Food, Nutrition and Health Research Group
Faculty of Education, Health & Community
IM Marsh Barkhill Road, Liverpool, L17 6BD
e: t.harrison@2015.ljmu.ac.uk<mailto:t.harrison@2015.ljmu.ac.uk>

[cid:image003.jpg@01D39293.3BDC0880]

Registered with the Association for Nutrition - www.associationfornutrition.org<http://www.associationfornutrition.org/>
Protecting the public and promoting high standards in evidence-based science and professional practice of nutrition.'

Important Notice: the information in this email and any attachments is for the sole use of the intended recipient(s). If you are not an intended recipient, or a person responsible for delivering it to an intended recipient, you should delete it from your system immediately without disclosing its contents elsewhere and advise the sender by returning the email or by telephoning a number contained in the body of the email. No responsibility is accepted for loss or damage arising from viruses or changes made to this message after it was sent. The views contained in this email are those of the author and not necessarily those of Liverpool John Moores University.

vincentvanhees, 2018-02-09:
based on issue #62 I consider this issue to be addressed and will now close it.

closed 2018-02-09 by vincentvanhees