wadpac/GGIR #62: Incomplete output files

TanjaHarrison, 2018-01-21:
Hi,
I have run my ActiGraph files through GGIR for parts 1 and 2 only. Everything seemed to go smoothly. There were no error messages.
However, in the 'Results' folder is only a 'QC' folder. When I open this there are no summary files or the pdf report. Only the 'Meta' folder contains files.

I attach the script that I have used. Would you mind having a look whether I did anything wrong here? Or let me know else might have gone wrong?

Thanks for your advice.
Tanja
[GGIR_activity_only_helpdesk20180121.pdf](https://github.com/wadpac/GGIR/files/1649624/GGIR_activity_only_helpdesk20180121.pdf)

vincentvanhees, 2018-02-09:
Do you see .RData files in the folders 'Z:/CALIBER_PA_Analysis/GGIR/GGIR_CALIBER_V1_output/meta/basic' and 'Z:/CALIBER_PA_Analysis/GGIR/GGIR_CALIBER_V1_output/meta/ms2.out'?

do.report =c(2) should be enough, the 4 is not needed because you are not performing sleep analyses

TanjaHarrison, 2018-02-19:
Hi,

Thank you for your reply.

I do have them in ‘'Z:/CALIBER_PA_Analysis/GGIR/GGIR_CALIBER_V1_output/meta/basic’ but NOT in 'Z:/CALIBER_PA_Analysis/GGIR/GGIR_CALIBER_V1_output/meta/ms2.out'.

Thanks,
Tanja

Important Notice: the information in this email and any attachments is for the sole use of the intended recipient(s). If you are not an intended recipient, or a person responsible for delivering it to an intended recipient, you should delete it from your system immediately without disclosing its contents elsewhere and advise the sender by returning the email or by telephoning a number contained in the body of the email. No responsibility is accepted for loss or damage arising from viruses or changes made to this message after it was sent. The views contained in this email are those of the author and not necessarily those of Liverpool John Moores University.

vincentvanhees, 2018-02-19:
That indicates that part 2 was not done, which may indicate that you did not run g.shell.GGIR with mode = 2.

TanjaHarrison, 2018-03-07:
Thank you for your advice. I have managed to run part 2 successfully now.

closed 2018-03-07 by TanjaHarrison