wadpac/GGIR #63: no results for part4 and part5

albacabre, 2018-01-30:
It seems that for some reason no output is generated for part4 and part5. 

Find below the errors that appear in the console, the script I am using, and the sessioninfo() information. If you need something else let me know Many thanks

------Console---------------------------------------------------------------------
Report part 4
 loading all the milestone data from part 4 this can take a few minutes
[1] "report not stored, because no results available"

_______________________________________________________________________________________________________________
Report part 5
 loading all the milestone data from part 5 this can take a few minutes
 generating output for every parameter configurations...
 NA-NA-NA-NA-NAreport not stored, because no results available NA-NA-NA-NA-report not stored, because no results available NA-NA-NA--NAreport not stored, because no results available NA-NA-NA--report not stored, because no results available NA-NA--NA-NAreport not stored, because no results available NA-NA--NA-report not stored, because no results available NA-NA---NAreport not stored, because no results available NA-NA---report not stored, because no results available NA--NA-NA-NAreport not stored, because no results available NA--NA-NA-report not stored, because no results available NA--NA--NAreport not stored, because no results available NA--NA--report not stored, because no results available NA---NA-NAreport not stored, because no results available NA---NA-report not stored, because no results available NA----NAreport not stored, because no results available NA----report not stored, because no results available -NA-NA-NA-NAreport not stored, because no results available -NA-NA-NA-r... <truncated>
_______________________________________________________________________________________________________________
Generate visual reports
[1] "File Sab14.bin.bin"
Error in xy.coords(x, y, xlabel, ylabel, log) : 
  'x' and 'y' lengths differ
In addition: Warning message:
In rm(outputfinal, outputfinal2) : object 'outputfinal2' not found
----------------------------------------------------------------------------------
-----------------------------------------------------------------------------------------

-----------Script----------------------------------

mode= c(1,2,3,4,5)
datadir= c("C:/Alba/03_Data/GeneActiv_Sab11y")     
outputdir= c("C:/Alba/03_Data/GeneActiv_Sab11y_results_sleeplog") 
f0 = 1
f1 = 3

g.shell.GGIR(#-------------------------------             
             # General parameters             
             #-------------------------------             
             mode=mode,             
             datadir=datadir,             
             outputdir=outputdir,             
             studyname=studyname,             
             f0=f0,             
             f1=f1,             
             overwrite = F,             
             do.imp=TRUE,             
             idloc=1,             
             print.filename=FALSE,             
             storefolderstructure = FALSE,             
             #-------------------------------
               # Part 1 parameters:             
             #-------------------------------
              windowsizes = c(5,900,3600),             
             do.cal=TRUE,             
             do.enmo = TRUE,             
             do.anglez=TRUE,             
             chunksize=1,             
             printsummary=TRUE,             
             #-------------------------------
             # Part 2 parameters:             
             #-------------------------------             
             strategy = 1,             
             ndayswindow=c(7,9),             
             hrs.del.start = 1,             
             hrs.del.end = 1,             
             maxdur = 9,             
             includedaycrit = 1,             
             L5M5window = c(0,24),             
             M5L5res = 10,             
             winhr = c(5,10),             
             qlevels = c(c(1380/1440),c(1410/1440)),             
             qwindow=c(0,24),             
             ilevels = c(seq(0,400,by=50),8000),             
             mvpathreshold =c(100,120),             
             mvpadur =c(1,5,10),             
             #-------------------------------
             # Part 3 parameters:             
             #-------------------------------             
             timethreshold= c(5),             
             anglethreshold=5,             
             ignorenonwear = TRUE,
             #-------------------------------
            # Part 4 parameters:             
             #-------------------------------             
             excludefirstlast = FALSE,             
             includenightcrit = 5,             
             def.noc.sleep = c(),             
             loglocation= "C:/Alba/03_Data/sleeplog_example.csv",             
             coln1 = 2,             
             colid=1,             
             outliers.only = FALSE,             
             criterror = 4,
             relyonsleeplog = FALSE,             
             sleeplogidnum = FALSE,             
             do.visual = TRUE,             
             nnights = 9,             
             visualreport = TRUE,             
             #-------------------------------             
             # Part 5 parameters:             
             #-------------------------------                        
             threshold.lig = c(30,40,50),             
             threshold.mod = c(100,120),             
             threshold.vig = c(400,500),             
             boutcriter = 0.8,             
             boutcriter.in = 0.9,             
             boutcriter.lig = 0.8,             
             boutcriter.mvpa = 0.8,             
             boutdur.in = c(10,20,30),             
             boutdur.lig = c(1,5,10),             
             boutdur.mvpa = c(1,5,10),             
             timewindow = c("WW"),
             #-----------------------------------             
             # Report generation             
             #-------------------------------
              do.report=c(1,2,3,4,5))
-------------------------------------------------------------------------------------------
-----------Sessioninfo----------
> sessionInfo()
R version 3.3.2 (2016-10-31)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows >= 8 x64 (build 9200)

locale:
[1] LC_COLLATE=Spanish_Spain.1252  LC_CTYPE=Spanish_Spain.1252    LC_MONETARY=Spanish_Spain.1252 LC_NUMERIC=C                  
[5] LC_TIME=Spanish_Spain.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] GGIR_1.5-16

loaded via a namespace (and not attached):
[1] zoo_1.7-14        tools_3.3.2       Rcpp_0.12.10      grid_3.3.2        GENEAread_1.1.1   data.table_1.10.2 bitops_1.0-6     
[8] lattice_0.20-34

vincentvanhees, 2018-02-09:
Thanks for posting this. To get a better understanding of the situation, can you please answer the following questions:
1. Do all your files have .bin.bin extension? This might be problematic for GGIR (I am not entirely sure). Can you try removing one .bin?
2. How does the output of part 2 look like, is it what you expect?
3. Based on the output of part2, can you confirm that the accelerometer was worn in the first 24 hours of the first participant?
4. Can you share your "C:/Alba/03_Data/sleeplog_example.csv", or at least the first couple of rows?

albacabre, 2018-02-09:
Hi Vincent,

Many thanks for your check. I tried it again only with one bin file 

1. I did it and nothing changed (the file was previously named 14.bin and now is 14)
2. Find attached what I got as part2 output and a screenshot of myresults folder 
3. Yes, participants wore the accelerometer since the beginning as it was placed by field nurses.
4. Find attached the sleeplog_example

[part2_daysummary.xlsx](https://github.com/wadpac/GGIR/files/1711049/part2_daysummary.xlsx)
[part2_summary.xlsx](https://github.com/wadpac/GGIR/files/1711050/part2_summary.xlsx)
[visualisation_sleep.pdf](https://github.com/wadpac/GGIR/files/1711051/visualisation_sleep.pdf)
[sleeplog_example.xlsx](https://github.com/wadpac/GGIR/files/1711053/sleeplog_example.xlsx)
[image]

Many thanks and do not hesitate to ask for further information if needed

Alba

vincentvanhees, 2018-02-09:
part2 looks good with over 7 days of measurement and wear duration.
maybe inspecting 'QC/plots to check data quality 1.pdf' would be a good extra check to be sure.

Next step is to check that part3 worked well. Can you go to .../meta/sleep.qc and check that there are pdf files in here with meaningful plots of arm angle and sleep classification during the night?

albacabre, 2018-02-12:
Hi Vincent,

I had to leave the office on friday, I will check this tomorrow when I am
in the office again. In case there are, I will send these files to you
through github

Thanks

Alba

-- 
Alba Cabré
Predoctoral researcher
IS*Global*
Barcelona Institute for Global Health - Campus MAR
Barcelona Biomedical Research Park (PRBB)
Doctor Aiguader, 88
08003 Barcelona, Spain
*alba.cabre@isglobal.org <alba.cabre@isglobal.org>*

-- 

This message is intended exclusively for its addressee and may contain 
information that is CONFIDENTIAL and protected by professional privilege. 
If you are not the intended recipient you are hereby notified that any 
dissemination, copy or disclosure of this communication is strictly 
prohibited by law. If this message has been received in error, please 
immediately notify us via e-mail and delete it.

DATA PROTECTION. We inform you that your personal data, including your 
e-mail address and data included in your email correspondence, are included 
in the ISGlobal Foundation files. Your personal data will be used for the 
purpose of contacting you and sending information on the activities of the 
above foundations. You can exercise your rights of access, rectification, 
cancellation and opposition by contacting the following address: 
lopd@isglobal.org. ISGlobal Privacy Policy at www.isglobal.org.

-----------------------------------------------------------------------------------------------------------------------------

CONFIDENCIALIDAD. Este mensaje y sus anexos se dirigen exclusivamente a su 
destinatario y puede contener información confidencial, por lo que la 
utilización, divulgación y/o copia sin autorización está prohibida por la 
legislación vigente. Si ha recibido este mensaje por error, le rogamos lo 
comunique inmediatamente por esta misma vía y proceda a su destrucción.

PROTECCIÓN DE DATOS. Sus datos de carácter personal utilizados en este 
envío, incluida su dirección de e-mail, forman parte de ficheros de 
titularidad de la Fundación ISGlobal  para cualquier finalidades de 
contacto, relación institucional y/o envío de información sobre sus 
actividades. Los datos que usted nos pueda facilitar contestando este 
correo quedarán incorporados en los correspondientes ficheros, autorizando 
el uso de su dirección de e-mail para las finalidades citadas. Puede 
ejercer los derechos de acceso, rectificación, cancelación y oposición 
dirigiéndose a lopd@isglobal.org . Política de privacidad en 
www.isglobal.org.

albacabre, 2018-02-13:
Hi Vincent,

Here the file 'QC/plots to check data quality 1.pdf' and the pdf with the plots of arm angle

[plots_to_check_data_quality_1.pdf](https://github.com/wadpac/GGIR/files/1720483/plots_to_check_data_quality_1.pdf)

[graphperday_id_14.bin_14.bin.pdf](https://github.com/wadpac/GGIR/files/1720486/graphperday_id_14.bin_14.bin.pdf)

I think they look good, maybe the problem is in step 4...let me know which is the next check

Many thanks,
Alba

vincentvanhees, 2018-02-13:
That looks good indeed, maybe something goes wrong in reading the sleeplog data and maybe it has to do with the fact that you only have one sleep night in your sleeplog file.
To test whether this is the problem: Can you copy paste the row with sleeplog_example.csv and paste it a row lower and give it a random other ID number? In that way GGIR may better be able to find the sleeplog data it is looking for. If this is the case, then some improvements need to be made in how GGIR handles sleeplog files with a single row of data.

albacabre, 2018-02-13:
I did it and the same error came out (run the script for one bin file (id=14)  and the sleeplog with two ids (14 and a random id)

Report part 4
 loading all the milestone data from part 4 this can take a few minutes
[1] "report not stored, because no results available"

______________________________________________________________________________________________________________________________________________
Report part 5
 loading all the milestone data from part 5 this can take a few minutes
 generating output for every parameter configurations...
 NA-NA-NA-NA-NAreport not stored, because no results available NA-NA-NA-NA-report not stored, because no results available NA-NA-NA--NAreport not stored, because no results available NA-NA-NA--report not stored, because no results available NA-NA--NA-NAreport not stored, because no results available NA-NA--NA-report not stored, because no results available NA-NA---NAreport not stored, because no results available NA-NA---report not stored, because no results available NA--NA-NA-NAreport not stored, because no results available NA--NA-NA-report not stored, because no results available NA--NA--NAreport not stored, because no results available NA--NA--report not stored, because no results available NA---NA-NAreport not stored, because no results available NA---NA-report not stored, because no results available NA----NAreport not stored, because no results available NA----report not stored, because no results available -NA-NA-NA-NAreport not stored, because no results available -NA-NA-NA-r... <truncated>
______________________________________________________________________________________________________________________________________________
Generate visual reports
Warning message:
In rm(outputfinal, outputfinal2) : object 'outputfinal2' not found

[sleeplog_example.xlsx](https://github.com/wadpac/GGIR/files/1720592/sleeplog_example.xlsx)

vincentvanhees, 2018-02-13:
Do you use that specific xlsx file? Can you convert it to csv, because that is how I designed GGIR.

albacabre, 2018-02-13:
Yes, .csv file but here I enclosed the excel file due to it does not allow to attach csv