wadpac/GGIR #71: Part 4 error when re-running Parts 2-4

jmkurka83, 2018-02-21:
Hi Vincent,
I am re-running parts 2-4 on a batch of 915 files, each with about 7 days+nights worth of data. The results output looks normal, however R produced this error
`Error in 1:nrow(spo) : argument of length 0`
Running the traceback() function simply printed the g.part4 function elements but didn't identify a line of code. What are your thoughts?
Thank you!
Jonathan

vincentvanhees, 2018-03-13:
I am working on a critical revision of error handling in g.part4 and will incorporate this in the next release of GGIR later this month, hopefully that will address your issue as well.

vincentvanhees, 2018-04-21:
This should be addressed now here in the version of GGIR on github, and an update to the version on CRAN should hopefully follow in a couple of days, look out for version 1.5-20 (or higher).
Thanks for reporting your issue and let us know if you experience any additional problems.

closed 2018-04-21 by vincentvanhees