wadpac/GGIR #73: Problems with sleep log

Lhurter, 2018-02-27:
Hi Vincent, 

I'm trying to run parts 1-5, with a sleep log. However, I keep getting the error message: 

Error in file(file, "rt") : cannot open the connection In addition: Warning message: In file(file, "rt") : cannot open file 'E:/STUDY_2/sleeplog1to8': No such file or directory

I made a sleep log as described in the instructions (.csv) and specified the loglocation correctly. Can you think of a reason why it can't open my log? Do you perhaps have an example of a sleep log, that I can use (by just adding my own sleep and wake times)? 

Thanks, 
Liezel

Lhurter, 2018-02-27:
No worries, I figured out what the problem was!

closed 2018-02-27 by Lhurter

Schmay00, 2022-06-06:
Hi Liezel

I'm currently running into the same issue. Would you mind sharing your solution?

Thanks
Maya