wadpac/GGIR #74: error in part.5: Error in if (timebb[s0] != as.character(timebb[s0])) { : missing...

dambach, 2018-02-27:
Hi everyone,

GGIR_1.5-17 gives an error when doing part 5 (never observed  with previous versions and no change in dataset).
It persists after regenerating metafiles for part 1,2,3,4:

____________________________
Part 5
 1Error in if (timebb[s0] != as.character(timebb[s0])) { : 
  missing value where TRUE/FALSE needed

Anyone encountered this issue ?

(I don't use sleep log entry)

Thanks,

Damien

dambach, 2018-03-08:
With def.noc.sleep = c(21,9) and since in this case the experiment start atmidnigth,  excludefirstlast must be set to TRUE.
This solved this issue.

vincentvanhees, 2018-03-17:
This should be fixed in the upcoming GGIR version 1.5-18. Let me know if the problem persists.

vincentvanhees, 2018-04-21:
Sorry to hear that. I hope it is addressed now here in the version of GGIR on github, and an update to the version on CRAN should hopefully follow in a couple of days, please look out for version 1.5-20 (or higher).
Thanks for reporting your issue and let us know if you experience any additional problems.

closed 2018-04-21 by vincentvanhees

JianWang2016, 2018-07-16:
Hi, Not sure if anyone else experiences this, I still get the error running part 5 with version 1.5-24.

 1Error in if (timebb[s0] != as.character(timebb[s0])) { : 
  missing value where TRUE/FALSE needed

> sessionInfo()
R version 3.5.0 (2018-04-23)
Platform: x86_64-pc-linux-gnu (64-bit)
Running under: Red Hat Enterprise Linux

Matrix products: default
BLAS: /opt/R/R-3.5.0/lib64/R/lib/libRblas.so
LAPACK: /opt/R/R-3.5.0/lib64/R/lib/libRlapack.so

locale:
 [1] LC_CTYPE=en_US.UTF-8       LC_NUMERIC=C               LC_TIME=en_US.UTF-8        LC_COLLATE=en_US.UTF-8    
 [5] LC_MONETARY=en_US.UTF-8    LC_MESSAGES=en_US.UTF-8    LC_PAPER=en_US.UTF-8       LC_NAME=C                 
 [9] LC_ADDRESS=C               LC_TELEPHONE=C             LC_MEASUREMENT=en_US.UTF-8 LC_IDENTIFICATION=C       

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] GGIR_1.5-24

loaded via a namespace (and not attached):
 [1] Rcpp_0.12.17      lattice_0.20-35   zoo_1.8-2         packrat_0.4.8-1   digest_0.6.15     rprojroot_1.3-2  
 [7] grid_3.5.0        jsonlite_1.5      backports_1.1.2   magrittr_1.5      evaluate_0.10.1   stringi_1.2.3    
[13] data.table_1.11.4 rmarkdown_1.10    tools_3.5.0       stringr_1.3.1     yaml_2.1.19       compiler_3.5.0   
[19] base64enc_0.1-3   htmltools_0.3.6   knitr_1.20

vincentvanhees, 2018-07-25:
Hi Jian, I have finally discovered what is causing this error. I could only reproduce it by changing my local timezone settings to the US, which indicated a timezone sensitive bug. I found a fix for it and will include it in the next version of GGIR, which I will upload to CRAN sometime in the upcoming week. Vincent