wadpac/GGIR #75: 'starttime0' not found

Drew74124, 2018-03-09:
Hi, 
I keep getting the error code below.  Any ideas?

Part 1
[1] "P1 file1"
[1] "---------------------------------------------"
[1] "investigate calibration of the sensors..."

Loading block: 1 2 3 4 5 6 7[1] "get meta data..."

Loading block: 1
Error in g.getstarttime(datafile = datafile, P = P, header = header, mon = mon,  : 
  object 'starttime0' not found

vincentvanhees, 2018-03-13:
This sounds like the timestamp format is not recognized. What is the format of the date-timestamp in the header of your data file? Do I understand it correct that you are working with an Actigraph file?

Drew74124, 2018-03-13:
Thanks for the reply Vincent,

------------ Data File Created By ActiGraph GT3X+ ActiLife v6.13.3 Firmware v3.2.1 date format d/MM/yyyy at 30 Hz  Filter Normal -----------

Serial Number: NEO1F07120358

Start Time 00:00:00

Start Date 14/02/2017

Epoch Period (hh:mm:ss) 00:00:00

Download Time 16:30:15

Download Date 22/02/2017

Current Memory Address: 0

Current Battery Voltage: 4.08     Mode = 12

Yes, Actigraph files.

The data will run using these CSV files if I use GGIR version 1.5-9, but not with the current version.

Regards
Drew

Dr Andrew Miller
(PhD Human Physiology)
Senior Lecturer
School of Education
University of Newcastle, Australia
EN2.05 Chittaway Rd,
OURIMBAH, NSW, 2258
Ph - 02 434 84214
Mob - +61 425 308 186

vincentvanhees, 2018-03-17:
Should be fixed in the code here on GitHub, and will also be included in the next release on CRAN in the upcoming days.

closed 2018-03-17 by vincentvanhees

Drew74124, 2018-03-19:
Thank you Vincent

Dr Andrew Miller
(PhD Human Physiology)
Senior Lecturer
School of Education
University of Newcastle, Australia
EN2.05 Chittaway Rd,
OURIMBAH, NSW, 2258
Ph - 02 434 84214
Mob - +61 425 308 186

commit 829bebe6 (2018-03-17): Fixes #54 and Fixes #58 and fixes #75
commit 3ff83b8c (2020-06-18): Fixes #54 and Fixes #58 and fixes #75