wadpac/GGIR #77: g.part5 "Error in if (ncol(output) > lastcolumn) { : argument is of length zero"

HeddaN, 2018-03-12:
Hi.
We're attempting to run GGIR on our actigraph data, but are having difficulties completing stage 5.
stages 1-4 are working well, but there seems to be an issue with lines 861 and 862 in `g.part5`.

```
        lastcolumn = which(colnames(output) == "bout.metric")
        if (ncol(output) > lastcolumn) {
          emptycols = which(output[1, ] == "" & output[2,
                                                       ] == "")
```

From what we can tell,  it's not finding the column named "bout.metric" in the output file, and thus the if below is failing, as it is trying to compare NA to `ncol(output)`. 

```
g.shell.GGIR(#=======================================
             # INPUT NEEDED:
             mode=c(5),
             datadir="M:/pc/dokumenter/GeneActivData1",
             outputdir="M:/pc/dokumenter/GeneActivData1",
             f0=1, f1=2,
             overwrite=TRUE,

             #-------------------------------
             # Part 1:
             #-------------------------------
             # Key functions: reading file, auto-calibration, and extracting features
             do.enmo = TRUE,             do.anglez=TRUE,
             chunksize=1,                printsummary=TRUE,
             #-------------------------------
             # Part 2:
             #-------------------------------
             strategy = 1,               ndayswindow=2,
             hrs.del.start = 0,          hrs.del.end = 0,
             maxdur = 9,                 includedaycrit = 16,
             winhr = c(5,10),
             qlevels = c(c(1380/1440),c(1410/1440)),
             qwindow=c(0,24),
             ilevels = c(seq(0,400,by=50),8000),
             mvpathreshold =c(100,120),
             bout.metric = 4,
             closedbout=FALSE,
             #-------------------------------
             # Part 3:
             #-------------------------------
             # Key functions: Sleep detection
             timethreshold= c(5),        anglethreshold=5,
             ignorenonwear = TRUE,
             #-------------------------------
             # Part 4:
             #-------------------------------
             # Key functions: Integrating sleep log (if available) with sleep detection
             # storing day and person specific summaries of sleep
             # excludefirstlast = TRUE,
             # includenightcrit = 16,
             # def.noc.sleep = c(),
             # loglocation= "M:/pc/dokumenter/GeneActivData1/sleeplog.csv",
             # outliers.only = TRUE,
             # criterror = 4,
             # relyonsleeplog = FALSE,
             # sleeplogidnum = TRUE,
             # colid=1,
             # coln1=2,
             # do.visual = TRUE,
             # nnights = 9,
             #-------------------------------
             # Part 5:
             # Key functions: Merging physical activity with sleep analyses
             #-------------------------------
             threshold.lig = c(30), threshold.mod = c(100),  threshold.vig = c(400),
             boutcriter = 0.8,      boutcriter.in = 0.9,     boutcriter.lig = 0.8,
             boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig = c(1,10),
             boutdur.mvpa = c(1),   timewindow = c("WW"),
             #-----------------------------------
             # Report generation
             #-------------------------------
             # Key functions: Generating reports based on meta-data
             do.report=c(2,4,5),
             visualreport=TRUE,     dofirstpage = TRUE,
             viewingwindow=1
             )
```

```

___________________________________________________________________________________________________________________________________
Part 5
 1Error in if (ncol(output) > lastcolumn) { : argument is of length zero

```

Session info is as follows:
```
R version 3.4.3 (2017-11-30)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 7 x64 (build 7601) Service Pack 1

Matrix products: default

locale:
[1] LC_COLLATE=Norwegian (Bokmål)_Norway.1252  LC_CTYPE=Norwegian (Bokmål)_Norway.1252   
[3] LC_MONETARY=Norwegian (Bokmål)_Norway.1252 LC_NUMERIC=C                              
[5] LC_TIME=Norwegian (Bokmål)_Norway.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] GGIR_1.5-17

loaded via a namespace (and not attached):
 [1] zoo_1.8-1           compiler_3.4.3      tools_3.4.3         yaml_2.1.17         Rcpp_0.12.15        grid_3.4.3         
 [7] GENEAread_1.1.1     data.table_1.10.4-3 bitops_1.0-6        lattice_0.20-35   
```

HeddaN, 2018-03-12:
We're uncertain what exactly is the need for the part in that `if`, but perhaps adding an `!is.na(lastcolumn) &` within the if might help?

Or perhaps you have an idea hy the column is not being generated?

vincentvanhees, 2018-03-13:
How long is the measurement you are trying to process with part5? If it is too short then part5 may struggle to generate a summary report. If this is the problem then we may need to add an extra check to the code for measurement duration to be used for generating a more informative error message.

HeddaN, 2018-03-14:
Thank you for your quick reply! We tried with one measurement that was 2 full nights and 1,5 day in duration, and one measurement that was 3 full days and 3 full nights in duration. Part 5 did not work with any of these. What is the minimum duration required for part5 to be able to generate complete output?

vincentvanhees, 2018-04-21:
This should be addressed now here in the version of GGIR on github, and an update to the version on CRAN should hopefully follow in a couple of days, look out for version 1.5-20 (or higher).
Thanks for reporting your issue and let us know if you experience any additional problems.

closed 2018-04-21 by vincentvanhees