wadpac/GGIR #81: Day, night and nightandday windows inconsistencies in g.part5

jhmigueles, 2018-04-06:
1. In the output data frame calculated in g.part5, the difference between acc_wake and acc_onset should be equal to bedtime (=dur_night_min), but it doesn't happen.

2. Wear time calculated from nonwear_perc and dur_day_min also presents inconsistencies. When calculating wear time from day + wear time from night, it is not the same that wear time from the complete nightandday.

vincentvanhees, 2018-04-08:
Thanks for spotting this. If you could investigate first that would be great.

Did you observe this when part5 was run with windows defined from midnight to midnight  (MM) or also for windows defined from waking up to waking up (WW)? Let's double check that it works correctly in both window setting before we consider this issue addressed.

jhmigueles, 2018-04-09:
I have been looking at this. I will do a pull request this same afternoon so you can check the changes that I propose to solve that.

Yes, I observed that for both MM and WW timewindows. In regards to the difference between acc_wake and acc_onset, the corresponding data was from the next row in all the cases I analyzed (i.e., if my dur_night_min is 480 for the first day, so the diff between acc_wake and acc_onset in the day 2 was the same). At least, it is happening when the firs day of measurement is of interest, for example when the accelerometer is initialized during the morning and this first day is of interest in the study. 

I have checked the code with some changes and I get consistent information for both MM and WW windows. But I agree, it would be great if you can double check it.

In regards to nonwear time, I just removed the round() function in the calculation and it seems to provide consistent data (i.e., wear_day + wear_night = wear_nightandday).

vincentvanhees, 2018-04-21:
Regarding item 1: Please note that in the latest version of GGIR (release 1.5-20) the variables in the MM output per day show the information per 24 hours between midnight and midnight. Therefore, the difference between acc_wake and acc_onset should not reflect 'sleep period time' or as we call it in the code 'dur_night_min', because acc_wake corresponds to the night on which the day starts and acc_onset corresponds to the night in which the day ends. So, I have checked that it does that.

Regarding item 2: This seems to be addressed now.

closed 2018-04-21 by vincentvanhees