wadpac/GGIR #86: Improve output variable dictionary for various parts

vincentvanhees, 2018-04-16:
At the moment output variables are explained in the manual and to some extend in the vignette, but considering the large number of analysis specific variables these overviews are at risk of being incomplete. We need a more systematic way of communicating what each variable stands for.

yaojiali, 2018-04-20:
Hi Vincent,

Thanks a lot for bringing this up. I realized that there might be a typo on page 49 of GGIR reference manual (https://cran.r-project.org/web/packages/GGIR/GGIR.pdf). 
If I understood correctly, TIN = Total inactivity = TOIN + TSIB, instead of  Total inactivity = TOIN + SIB. (sceenshot attached). If possible, could you kindly clarify? 
[image]

Many thanks for your time!
regards,
Jiali

vincentvanhees, 2018-04-21:
thank you Jaili, in the new version of the documentation this will be corrected.

Sustained inactivity bouts can be part of inactivity bouts, but do not have to.
TSIB are all SIB taken together.
SIB refers to all sustained inactivity bouts not part of inactivity bouts.

In the documentation I will explain this as:
Variable names with SIB in them are the sustained inactivity bouts that are
not part of the inactivity bouts. Only TSIB shows the total time spent in all
sustained inactivity bouts. I make this distinction, because sustained inactivity
bouts, can be part of inactivity bouts and are then counted towards the inactivity
bout (INB variables) and not towards the SIB category.

yaojiali, 2018-04-25:
Thank you very much for the clarification!
Regards,
Jiali

closed 2018-05-07 by vincentvanhees