wadpac/GGIR #87: Support for reading .gt3x files

TheTS, 2018-04-23:
Hi Vincent

Not an issue, but a question: 

I'm wondering if you plan to support reading the ActiGraph .gt3x file format? Similar to reading binary data from the axivity.

https://github.com/actigraph/GT3X-File-Format

I know GGIR currently supports reading ActiGraph csvs exported via Actilife, but with this function, you could bypass the Actilife software altogether. 

There is some MATLAB/Java/Python code around, but it would be nice to do this using R.

Cheers

Tom

vincentvanhees, 2018-04-25:
Hi Tom,
Good idea. At the moment I already have a lot of GGIR tasks on my plate. Possible ways forward may include:
- Someone develops this and I spend a few hours to help embed this within GGIR or help build a separate R package around it.
- Someone pays me to develop this code as a consultancy.
- Combination of both.
Cheers, Vincent

vincentvanhees, 2018-04-26:
Thanks, that would be great!

TheTS, 2018-04-29:
So just an update - I've put together some code which seems to return the same result as the gt3x to raw csv export in Actilife. It's not 100% complete as the checksum for each block should probably be checked to verify there were no errors. Also it won't work with the really old devices where the data are saved using a different structure.

I may have time to work on it more later, but I'll just dump the code here for now in case anyone wants to do something with it. 

Cheers

T

```r
//Example usage
data <- read_gt3x("test_file.gt3x")
```

C++ code:
```cpp
#include <Rcpp.h>

using namespace Rcpp;

//' Convert a vector of raw bytes to yxz g values
//'
//' Raw activity samples packed into 12-bit values in YXZ order.
//'
//' @param bytes A RawVector
//' @param scale Scale factor to return acceleration in g
//' @return A NumericVector of g values in yxz order
//'
// [[Rcpp::export]]
NumericVector read_activityC(RawVector bytes, double scale) {
  NumericVector res;

  int n = bytes.size() -1;
  bool alt = true;
  uint16_t data;

  for (int i=0; i < n; ++i) {
    if (alt) {
      data = (((bytes[i] << 8) | ((bytes[i+1]))) >> 4);
    } else {
      data = (((bytes[i] << 8) | ((bytes[i+1]))) & 0xFFF);
      i++;
    }
    alt = !alt;

    if (data > 0x7FF) data |= 0xF000;

    res.push_back(round(((int16_t)data) / scale * 1000.0f) / 1000.0f);
  }

  return res;
}
```
And the R code:

```r
#' Read acceleration and timestamp from a log.bin file
#'
#' Note, this passes a raw vector to c++ for parsing
#'
#' @param fid A file connection able to read binary
#' @param accScale Scale factor to return acceleration in g
read_acceleration <- function(fid, accScale) {
  x <- 1
  data <- list()
  time <- list()

  # First byte is the seperator
  while (length(readChar(fid, 1, useBytes = TRUE)) > 0) {

    # Record type - 1 byte
    type <- readBin(fid, integer(), size = 1)

    # Unix timestamp - 4 bytes
    ts <- readBin(fid, integer(), size = 4)

    # Size of payload (in bytes) - 2 bytes
    size <- readBin(fid, integer(), size = 2)

    # Type 0 is ACTIVITY
    if (type == 0L) {
      if (x==1) prevTs <- ts - 1

      # For missing samples - impute last yxz values
      if (ts - prevTs != 1)
        while (ts - prevTs != 1) {
          data[[x]] <- data[[x-1]]
          prevTs <- prevTs + 1
          time[[x]] <- prevTs
          x <- x + 1
        }

      data[[x]] <- read_activityC(readBin(fid, raw(), size), accScale)
      time[[x]] <- ts
      prevTs <- ts
      x <- x + 1

    } else {
      readChar(fid, size, useBytes = TRUE)
    }
    # Checksum - 1 byte
    readBin(fid, integer(), size = 1)
  }

  list(time = unlist(time), data = matrix(unlist(data), ncol = 3, byrow = TRUE))
}

#' Read the info.txt file inside a .gt3x
#' This contains key items to help parse activity data
#'
#' @param fid A file connection to the info.txt
read_info <- function(fid) {

  # Helper to convert .NET ticks to timestamp
  ticks_ts <- function(ticks) {
    as.POSIXct(ticks / 1e7, origin = '0001-01-01', tz = 'GMT')
  }

  info_file <- data.frame(key = readLines(fid)) %>%
    separate(key, c("key", "value"), ": ") %>%
    spread(key, value)
  info_file <- info_file %>%
    rename_all(funs(make.names(tolower(names(info_file))))) %>%
    mutate_at(vars(-device.type, -firmware, -serial.number,
                   -subject.name, -timezone), as.numeric) %>%
    mutate_at(vars(download.date, last.sample.time,
                   start.date, stop.date), ticks_ts)
}

#' Read raw data from .gt3x
#'
#' @param fileName Path to .gt3x file
#'
#' @export
read_gt3x <- function(fileName){

  info.con <- unz(fileName, "info.txt")
  log.con <- unz(fileName, "log.bin", open = "rb")

  info <- read_info(info.con)
  close(info.con)

  data <- read_acceleration(log.con, info$acceleration.scale)
  close(log.con)

  # Create 13-digit unix timestamp to preserve milliseconds
  time <- list()
  for (i in 1:(length(data$time)-1)) {
    time[[i]] <- seq(data$time[i] * 1e3, by = 1e3 / info$sample.rate,
                     length.out = info$sample.rate)
  }

  data <- data.frame(unlist(time), data$data[,c(2, 1, 3)])
  names(data) <- c("timestamp", "x", "y", "z")

  return(invisible(list(header = info, data = data)))
}
```
And the structure of the result:
```
> str(data)
List of 2
 $ header:'data.frame':	1 obs. of  18 variables:
  ..$ acceleration.max  : num 8
  ..$ acceleration.min  : num -8
  ..$ acceleration.scale: num 256
  ..$ battery.voltage   : num 4.19
  ..$ board.revision    : num 1
  ..$ device.type       : chr "wGT3XBT"
  ..$ download.date     : POSIXct[1:1], format: "2015-04-09 17:36:33"
  ..$ firmware          : chr "1.5.0"
  ..$ height            : num 183
  ..$ last.sample.time  : POSIXct[1:1], format: "2015-04-09 17:36:33"
  ..$ mass              : num 70.3
  ..$ sample.rate       : num 30
  ..$ serial.number     : chr "MOS2A45130448"
  ..$ start.date        : POSIXct[1:1], format: "2015-04-09 14:02:00"
  ..$ stop.date         : POSIXct[1:1], format: "2017-01-27 12:00:00"
  ..$ subject.name      : chr "shakertest-wGT3X-BT-2"
  ..$ timezone          : chr "-04:00:00"
  ..$ unexpected.resets : num 0
 $ data  :'data.frame':	385800 obs. of  4 variables:
  ..$ timestamp: num [1:385800] 1.43e+12 1.43e+12 1.43e+12 1.43e+12 1.43e+12 ...
  ..$ x        : num [1:385800] 0 -0.105 -0.031 -0.023 -0.02 -0.023 -0.027 -0.023 -0.02 -0.023 ...
  ..$ y        : num [1:385800] 0 0.105 0.172 0.172 0.168 0.16 0.168 0.168 0.168 0.16 ...
  ..$ z        : num [1:385800] 0 -0.48 -0.844 -0.852 -0.848 -0.855 -0.848 -0.855 -0.848 -0.844 ...
```

Mirkes, 2018-05-01:
> Thanks Tom! @Mirkes<https://github.com/Mirkes> would you be interested to have a look at this? It is for Actigraph data format<https://github.com/actigraph/GT3X-File-Format>. If not, I can also have a look at it in the upcoming months, but am quite busy at the moment.

Hi,

I’ll look this code next week. Unfortunately now I do not have time.

Best regards,
Evgeny

Mirkes, 2018-05-29:
Sorry for long delay.
I have read presented text. This text looks like absolutely correct.
Data format assumed 26 different types of data blocks. Proposed software read the type 0 only.
Some other types contains summary information but part of types can be useful. From the other side for GGIR we have to read the same information from each source.
Really there are two other formats of data representation in types 24-25 and in type 26.
Types 24-25 are not described yet.
Data from type 26 can be read (documentation is enough  detailed).
For old data file format is also descripted.

I cannot estimate probability to meat file of each sub format and cannot recognise which formats it is desirable to parse.

I hope it is useful,
Evgeny

jmkurka83, 2018-06-12:
Hi Tom,
I'm currently going through Actilife to convert raw gt3x to raw csv. I've noticed that the raw csv file doesn't have the ID in the header. Does your addition pick up an ID in the header of the gt3x file (if there is one)? The ID is important to us to match it to a sleep log.
Thanks!

TheTS, 2018-06-12:
Hi Jonathan

The gt3x file is basically a .zip that contains 2 files: the binary data file, and a text file containing other info like device settings. IIRC the participant ID is stored in the text file under 'subject name', as well as the serial number etc. I made it save all this info in the header data.frame. You should be able to see this in the example output above.

Cheers

T

Sent from phone

From: Jonathan Kurka
Sent: Wednesday, 13 June, 05:05
Subject: Re: [wadpac/GGIR] Support for reading .gt3x files (#87)
To: wadpac/GGIR
Cc: Tom Stewart, Author

Hi Tom,
I'm currently going through Actilife to convert raw gt3x to raw csv. I've noticed that the raw csv file doesn't have the ID in the header. Does your addition pick up an ID in the header of the gt3x file (if there is one)? The ID is important to us to match it to a sleep log.
Thanks!

TuomoNieminen, 2018-09-11:
Hello,

I work as a researcher at the National Institute for Health and Welfare, Finland. I have not yet used the GGIR package, but I have spent the past couple months working on this issue of reading the raw .gt3x files into R. 

I have coded an RCPP parser for the Actigraph .gt3x data by following the GT3X GitHub documentation. I have implemented the parser for both possible activity data formats (id's 0 and 26) and also the parameters (id 21).

My code is in an r-package here: https://github.com/THLfi/read.gt3x
The documentation of the package, including a vignette, is here: https://thlfi.github.io/read.gt3x
Note that I am a beginner with C++. My apologies if the code is messy.

It would be excellent if this feature could be added to the GGIR package. Changes to my package can be made where necessary if it helps in adding this feature to GGiR. The code can also be used and added to GGIR. Please let me know your thoughts.

vincentvanhees, 2018-09-12:
Thanks for sharing your work! This sounds great.

Do I understand it correctly that your code reads the data as one big block? If yes, would it be a lot of work to add two input argument to read.gt3x that specify the start and the end of time window to be read (row or page number rather than clock time is also fine). That ability would then enable loading large Actigraph files that do not fit into memory. See also Evgeny did it with function [g.readcwa](https://github.com/wadpac/GGIR/blob/master/R/g.cwaread.R) and how [GENEAread](https://cran.r-project.org/web/packages/GENEAread/index.html) did it for their data. GGIR uses this in [g.readaccfile](https://github.com/wadpac/GGIR/blob/master/R/g.readaccfile.R) to read the data in segments.

Regarding the embedding in GGIR:
Considering that your package will be used beyond GGIR it may be best to have it as a separate package on CRAN (or github might also work), which GGIR could then use as a dependency. This to avoid having one giant package for whom most people only use a fraction of the functionality.

But, we could also first focus on making your code work in GGIR and then migrate it to a separate package later on, possibly together with a collection of other functions for reading accelerometer data:
- Evgeny's g.cwaread in GGIR
- My g.binread for GENEA data in GGIR
- Your read.gt3x

Cheers, Vincent

TuomoNieminen, 2018-09-17:
Hello Vincent,

Thank you for the reply. You are correct that the read.gt3x function calls a C++ function, which processes the whole stream of binary data byte by byte, from the beginning to end, extracts the activity samples and then returns them as a single matrix. 

I can see implementing your suggestion in reasonable time, perhaps using row index and/or timestamp to specify the start and end points, since page number seems perhaps out of place for binary input data. I'll have a look at the other implementations, thank you for the links.

If I may, my suggestion for the integration between GGIR and data readers would be to formally define the data format that accelerometer readers should produce. I can see that g.readaccfile produces a list with meta + data. (read.gt3x currently includes needed metadata as attributes to the data matrix it returns)

To formalize the data structure, I think it would make sense to define an S3 class and then have the readers return an object of that class (It could, for example, be the already used 2-element list with certain required elements and then the class attribute "accelerometer" or something similar). That way it should be easier to develop code that uses accelerometer data from different devices. 

How does this sound?

vincentvanhees, 2018-09-24:
Sorry for the slow reply Tuomo, I think that is a very good suggestion. There are more data objects within GGIR that could benefit from a formal class definition. I will start working on it in the upcoming months, but because I also have other deadlines and progress may not be quick in the first instance.

TuomoNieminen, 2018-09-27:
No problem at all Vincent. Thank you for the reply. I understand that you are busy. That is the way of the world, right? Not enough time for the fun projects :)

I will start working on your suggested addition to the read.gt3x package of specifying a time period or indices to read. I will also aim to pass CRAN checks.

Meanwhile I might be tempted to use GGIR, so:  
Even if not formally defined, is there currently some R format for the data that read.gt3x could return which would be directly usable by GGIR functions? Or are some changes to GGIR necessary to allow usage of a new way to read data?

vincentvanhees, 2020-03-12:
Some time has passed since the above conversation in 2018.

I see @TuomoNieminen has been continuing the work on his [gt3x.read](https://github.com/THLfi/read.gt3x) package as did @paulhibbing on the [AGread](https://github.com/paulhibbing/AGread) package. Also I see that @muschellij2 is actively contributing to both projects.

It looks like both packages facilitate reading the raw gt3x data, but neither is able yet to read specific chunks to facilitate memory management as discussed above. Correct me if I am wrong! 

Are you currently still aiming to develop such functionality?
- If yes, what is your time plan and how can I be of help?
- If not, what are the obstacles (time? funding? expertise?).

Note: For other binary formats we have in GGIR the option to read the file in chunks, e.g. [g.binread](https://github.com/wadpac/GGIR/blob/master/R/g.binread.R) and [g.cwaread](https://github.com/wadpac/GGIR/blob/master/R/g.cwaread.R). Maybe this provides some ideas about how to do it.

muschellij2, 2020-03-12:
So my assessment is that `read.gt3x` can read in both old (NHANES) and new gt3x formats, whereas `AGread` cannot.  `read.gt3x` is also much faster in this regard currently .   

## Full data
`AGread` can read in more of the blocks of the data (such as `BATTERY` and `LUX` fields) whereas `read.gt3x` only reads `ACTIVITY` data.  `AGread` cannot read in `ACTIVITY2` data I believe (or regular `ACTIVITY` blocks, I cannot remember which), whereas `read.gt3x` can.  

## Timings and Imputation
There are some issues with the timings on the device (other than the normal timezone hell).  I have noted that there is something off as noted in https://github.com/THLfi/read.gt3x/issues/13.  I don't know what it is, but my gut thinks it may be some overflow problem or initialization of an origin or timezone.   Imputing `zeros` or putting in the repeated value if nothing is recorded (battery saving mode), is also difficult and there seems to be some issues there.  Haven't explored enough to assess fully.

vincentvanhees, 2020-03-12:
thanks @muschellij2, very helpful. I had a play with read.gt3x today, indeed very fast

The 1 week of example data that comes with the package has an object.size() in R of 2 Gb, which my virtual machine with 4 CPU and 8GB was able to handle as long as I did not have too many other processes going on. 

**My original concern:**
My original concern was that object size for a week worth of data would not leave sufficient memory for other R objects in the process of analysing the data. This concern originates from when I was working with 7 column GENEActiv data around 2010. The PC I was working on couldn't hold a week of it in memory and that is why I went down the road of loading data in chunks.

**My plan for now:**
I will try to embed gt3x.read in a development branch of GGIR in the upcoming weeks. Hopefully that will tell us whether memory management is still a concern. If it is then removing timestamps could be a way to reduce memory. If we keep a copy of the first timestamp and the sample rate then timestamps can be recreated at a later point.

paulhibbing, 2020-03-13:
A couple notes wrt `AGread`:

* It is possible to read only certain packet types (e.g. ACTIVITY2), by specifying which packets you want via the `include` argument. But unless you're skipping accelerometer or IMU packets, the functionality probably causes a net slowdown. Basically the whole file gets scanned first to determine where each packet starts/stops, and what the packet types are. Then those records get sorted and filtered according to what's specified in `include`. All of that happens before any packets are actually parsed. Skipping small packets like BATTERY and CAPSENSE probably doesn't save enough time to justify scanning the whole file first. Maybe that can be avoided through a streaming approach, but that's beyond me at this point. (And there's no way around that scanning process, so it's a bottleneck whether or not you want to customize anything... That's a design flaw.) For the record, I don't think memory or runtime would be greatly impacted by including vs excluding certain packet types (except for accelerometer or IMU packets) since most of them are very light. 

* I'm guessing other slowdowns occur because of the ways I've handled processes like interpolation and midpoint rounding. My goal has been to give output just as you would get in a `RAW.csv` or `IMU.csv` file, so I've had to add in some of those extra processes and likely done so in time-inefficient ways. (My version of mid-point rounding is painful to look at, but I just don't have the knowhow to do it elegantly, or the time to figure it out.)

* I've encountered several pitfalls with things like interpolation and idle sleep mode, which I've only been able to sort out through direct communication with the folks at ActiGraph. Some of those things should show pretty clearly in the `AGread` history (both commits and issues), and I think there are some that can have pretty big implications. Lots to be wary about...

* Idle sleep mode is a big consideration for how timestamps are going to be calculated. If you're aiming to save time/memory by waiting til after downsampling to calculate timestamps, don't forget that there can be considerable time gaps between packets. I don't know that timestamp calculation is efficient in `AGread` (probably not, if the other processes are any indication), but it is robust to this issue.

vincentvanhees, 2020-03-13:
Thanks Paul. It sounds like you dedicated a lot of time to quality reassurance. So, ideally I would like to facilitate GGIR usage with both your and Tuomo's package. 

I want GGIR to work on even simple standard laptops, because that is the reality in most research labs. When I try to use  AGRead::read_gt3x in a 4CPU 8GB virtual machine to read the example file from Tuomo's read.gt3x package (EE_left_29.5.2017-05-30.gt3x) I get `Error: cannot allocate vector of size 2.1 Gb`. If processing a single file including overhead can be done under 2GB then it becomes possible to process multiple files in parallel on more powerful machines. This is what I currently do for all other data formats. 

From a GGIR perspective, possible ways forward:
- I give up on GGIR users with less than 16GB RAM who want to process their gt3x data, which makes it possible to facilitate embedding of both your packages.
- I try to help adapt your code to read the file in chunks. I have a feeling that this may be a lot of work for which I probably am not well skilled.
- For the moment I try to get read.gt3x to work to explore how far I get with that.

**Timestamps irregularity:** this may be addressed by resampling the data with the available timestamps before throwing them away. As part of this step it may even be worth considering bringing the sample rate down to 50 Hertz because for many applications 100 Hertz is not essential for the things GGIR does.

muschellij2, 2020-03-13:
I would agree about 100hz but that’s what’s mims units need (or resample
to) and the next NHANES iteration are using them so I’m guessing they are
going to get more popular

TuomoNieminen, 2020-05-03:
Hello,

Apologies for the late reply, my focus has been on COVID-19 related work for the past couple months.

"
It looks like both packages facilitate reading the raw gt3x data, but neither is able yet to read specific chunks to facilitate memory management as discussed above. Correct me if I am wrong!

Are you currently still aiming to develop such functionality?

If yes, what is your time plan and how can I be of help?
If not, what are the obstacles (time? funding? expertise?).
"

Correct, this feature has not been implemented. Working with activity data is now a very small part of my current work and so the read.gt3x project is essentially not funded. Development occurs during my own time. It is therefore difficult to justify spending a lof of time on feature requests. I do want to maintain and fix bugs and am very grateful for feedback. I also don't mind helping out with the implementation of new features when I can.

vincentvanhees, 2020-05-04:
Thanks @TuomoNieminen for this update. I fully understand your time budget constraints.

I am in the process of implementing the ability for GGIR to read gunzipped .csv files, see #289. Although the file conversion steps are time consuming and require ActiLife, it at least addresses the storage size concern of .csv-exports and will be useful for other accelerometer brands too.

vincentvanhees, 2021-07-20:
I am closing this old issue to avoid the impression that I am working on it or plan to work on it. Feel free to re-open it if and when an R package becomes available that is able to read the .gt3x files in chunks. 

For now Tuomo's R package can be used to read .gt3x data on machines with sufficient memory to hold the data, which can then be stored in .csv format for use by GGIR or other software tools. 

@actigraph if you would like your .gt3x data format to be easily integrated in open source software tools like GGIR then please develop and share open source code for this. As you will see in this thread, the key requirement is that the data can be read in chunks to aid memory management. I am happy for GGIR to be compatible with all accelerometer brands, but I hope you understand that I cannot invest unpaid time in resolving brand-specific data reading issues.

closed 2021-07-20 by vincentvanhees