wadpac/GGIR #89: Part 4 Error - WKN2 > SLN2

jmkurka83, 2018-04-29:
Hello,
I've run part1 in small batches on 600+ files. I'm now trying parts2-4 and getting an error in regards to the sleep log.
> Part 4
Error in if (WKN2 > SLN2) { : missing value where TRUE/FALSE needed
`> traceback()
3: g.loadlog(loglocation, coln1, colid, nnights, sleeplogidnum)`.
The loglocation specification in part4 is correctly identifying a .csv file with the appropriate formatting.

Thoughts? Thank you!

vincentvanhees, 2018-05-07:
Hi Jonathan, I think there is a problem with the format in one or more of your sleeplog cells. It happens in [this g.loadlog function](https://github.com/wadpac/GGIR/blob/1b4e3ad0cdadac27a19e669d7e0877b0fe5d19cf/R/g.loadlog.R) which as you can see already checks for empty cells and missing values. You can use this g.loadlog function to investigate what cell in your sleeplog file is causing the problem and that should hopefully give you a clue on how to fix it.

jmkurka83, 2018-05-15:
Hi Vincent! I've gone through the code trying to troubleshoot my issue but have come to a deadend and could use your advice. Using the g.loadlog code, the first time the variable SL is created it is correct
`head(SL)
[1] "15:16:00" "02:00:00" "23:00:00" "1:00:00"  "23:45:00" "20:30:00"`
However, after it goes through this line of code `SL[j] = paste(SLN[1],":",SLN[2],":",SLN[3],sep="")` it looks like this:
`head(SL)
[1] "15:16:0" "2:0:0"   "23:0:0"  "1:0:0"   "23:45:0"
[6] "20:30:0"`
I know GGIR pulls apart the initial SL variable, checks for the seconds, then adds them if not found. I noticed that the code line `if (length(SLN) == 2) SLN = c(SLN,0)` only adds one 0. Could that be the issue? It's also worth noting that my sleep log does appear to have seconds. I've attached it as proof (saved as xlsx, I have a csv version for GGIR use). It's formatted identically to a sleep log that functions.
What do you think? Any help with these basics of R coding and GGIR functionality is appreciated!
[wave2_GGIRlog.xlsx](https://github.com/wadpac/GGIR/files/2003404/wave2_GGIRlog.xlsx)

jmkurka83, 2018-06-05:
My csv file contained spaces between the commas that were not viewable in Excel. Opened the file in notepad and manually removed the spaces.

vincentvanhees, 2018-07-12:
Hi Jonathan, your message from the 15th of May may be a lead to investigate. However, does your last message mean that the issue is resolved or not? I will re-open this issue now as a reminder to look into it.
Vincent

jmkurka83, 2018-08-03:
Hi Vincent!
Sorry for the delay. Yes, I fixed the issue by removing the spaces in the notepad .txt file, opening this in Excel, then saving as a csv file. Obviously the files looked identical in Excel, but the lack of the 'space' between the commas did the trick.
Thanks for checking in!