wadpac/GGIR #90: Missing output

ingrre, 2018-05-04:
Hi! 
I have tried to run GGIR ( 1.5-21) on a .bin-file (actigraph worn for 7 days). We don't have sleep log, so that's left out of the script. 
As an output we get all of part 2, part 4 (even though it is left out from the script) and two files for part 5 + Report_(name of file).pdf. However, the report from part 5 is incomplete as the visual summary of the physical activity and sleep patterns is empty and only for 1 day. Also, when running the script on new data, the new information is added in the already created files, but the Report_(name of file).pdf is not updated. 

Any thoughts on what is going on? 

The error message I get looks like this: 

Error in xy.coords(x, y, xlabel, ylabel, log) : 
  'x' and 'y' lengths differ
In addition: There were 50 or more warnings (use warnings() to see the first 50)
 
 > warnings()
Warning messages:
1: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
2: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
3: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
4: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
5: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
6: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
7: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
8: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
9: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
10: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
11: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
12: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
13: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
14: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
15: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
16: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
17: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
18: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
19: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
20: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
21: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
22: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
23: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
24: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
25: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
26: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
27: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
28: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
29: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
30: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
31: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
32: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
33: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
34: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
35: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
36: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
37: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
38: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
39: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
40: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
41: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
42: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
43: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
44: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
45: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
46: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
47: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
48: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
49: In tmp5[i] <- fnamesfull[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
50: In tmp6[i] <- phase[which(fnamesshort == ff)] :
  number of items to replace is not a multiple of replacement length
>

vincentvanhees, 2018-05-07:
1. Did you get a report from part 2? If yes, how did that look like?
2. May I assume you are using function g.shell.GGIR? If yes, what arguments did you use for g.shell.GGIR? 
3. The above error message might indicate that GGIR parts were run with different versions of GGIR and/or different arguments to g.shell.GGIR. Can you confirm that this is not the case?

ingrre, 2018-05-09:
Yes. We do get some output. We managed to get some feasible results before
the weekend, but it will not give us the same results again. We tried to
remove the .bin file that gave some results (last week) and tried to
analyze a new one. When transferring a new file to the folder in use by the
script, and running the script again, all output documents are modified
(judging by the date last modified), but the data from the current
actigraph-file is not seen in the actual result files. The output we get
(and which are not updated with the current file-information) is all of
part 2: *part2_daysummary.csv, part2_summary.csv,
QC/plots_to_check_data_quality_1.pdf and data_quality_report.csv. *The csv
files are excel-sheets with all variables in them, but only for two .bin
files, whereas we have a "file summary report" for 4 -bin-files now (we
have tried different files several times). The same goes for the quality
report: the result from the same two .bin-files are present there (but no
new one is made).

I am using GGIR_1.5-21. How is it possible to use different versions of the
GGIR-package?

This is the script we use:

g.shell.GGIR(#=======================================
             # INPUT NEEDED:
             mode=c(1,2,3,4,5),
             #mode=c(4,5),
             datadir="M:/pc/Dokumenter/GeneActive_R",
             outputdir="M:/pc/Dokumenter/GeneActive_R",
             f0=1, f1=2,
             overwrite=T,
             #-------------------------------
             # Part 1:
             #-------------------------------
             # Key functions: reading file, auto-calibration, and
extracting features
             do.enmo = TRUE,             do.anglez=TRUE,
             chunksize=1,                printsummary=TRUE,
             #-------------------------------
             # Part 2:
             #-------------------------------
             strategy = 2,               ndayswindow=7,
             hrs.del.start = 0,          hrs.del.end = 0,
             maxdur = 9,                 includedaycrit = 16,
             winhr = c(5,10),
             qlevels = c(c(1380/1440),c(1410/1440)),
             qwindow=c(0,24),
             ilevels = c(seq(0,400,by=50),8000),
             mvpathreshold =c(100,120),
             bout.metric = 4,
             closedbout=FALSE,
             #-------------------------------
             # Part 3:
             #-------------------------------
             # Key functions: Sleep detection
             timethreshold= c(5),        anglethreshold=5,
             ignorenonwear = TRUE,
             #-------------------------------
             # Part 4:
             #-------------------------------
             # Key functions: Integrating sleep log (if available) with
sleep detection
             # storing day and person specific summaries of sleep
             #excludefirstlast = TRUE,
             #includenightcrit = 16,
             #def.noc.sleep = c(),
             #loglocation= "C:/mydata/sleeplog.csv",
             #outliers.only = TRUE,
             #criterror = 4,
             #relyonsleeplog = FALSE,
             #sleeplogidnum = TRUE,
             #colid=1,
             #coln1=2,
             #do.visual = TRUE,
             #nnights = 9,
             #-------------------------------
             # Part 5:
             # Key functions: Merging physical activity with sleep analyses
             #-------------------------------
             threshold.lig = c(30), threshold.mod = c(100),  threshold.vig
= c(400),
             boutcriter = 0.8,      boutcriter.in = 0.9,     boutcriter.lig
= 0.8,
             boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig =
c(1,10),
             boutdur.mvpa = c(1),   timewindow = c("WW"),
             #-----------------------------------
             # Report generation
             #-------------------------------
             # Key functions: Generating reports based on meta-data
             do.report=c(2,4,5),
             visualreport=TRUE,     dofirstpage = TRUE,
             viewingwindow=1)

vincentvanhees, 2018-05-13:
Your problem description is unclear to me. Try to start from scratch, avoid moving files around, and describe the problem in a way such that it can be reproduced for investigation.

ingrre, 2018-05-22:
So I find that with most files the analysis now works well (thus it might
be an internal error in the files that do not respond correctly). However,
it does not seem seem like it is possible to analyze several files at the
same time. For now I generate a new folder for each analysis, where the
only file in the folder is the actigraph-file to be analyzed. Is this
familiar to you?

vincentvanhees, 2018-05-28:
I am not sure I understand. If you create one folder for all your data and specify its directory with 'datadir' it should just work.

closed 2018-05-28 by vincentvanhees