wadpac/GGIR #96: Error when specifying files using f0 and f1

jmkurka83, 2018-06-05:
Hi Vincent,

I just wanted to let you know there is still an error when trying to run only certain files within a folder. While specifying f0=11 and f1=13 part1 will choose the 11th and 13th file, but an error occurs in part 2:
`Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
In addition: Warning message:
In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file 'X:output_csvfiles/meta/basic/NA', probable reason 'No such file or directory'`
I've started running part1 with the above specifications. Then change them (in this case) to f0=1 and f1=3 (since there are only 3 files to run) for parts 2-4.
Thanks as always for your work on this!

vincentvanhees, 2018-07-12:
Hi Jonathan,

Thanks, f0 and f1 are indeed designed to be specific to the files available for each part. So, if part2 only finds two milestonefiles from part1 it will not be able to analyse the 11 till the 13th file.

I am not sure how to address this, because users are probably used to this behaviour, and do not want the software to be too clever and suddenly select different files.

We could let GGIR produce a more informative error message, because I understand that this is confusing. I am not sure though whether changing the behaviour of f0 and f1 is desirable.

Vincent

jmkurka83, 2018-08-03:
Hi Vincent,
I agree, the GGIR behavior makes sense. Our team has been working on a standard protocol on how to use GGIR in a "part-by-part" process, as this makes navigating the mass amount of data a bit easier. I'm sure it's not high on the list, but yes more informative error messages would be very helpful!
Thank you for your response and dedication to this project.

Jonathan