wadpac/GGIR #101: version 1.5-24 incorrect reporting of valid days

Alex-Row, 2018-07-12:
Hi Vincent,

I’m re-processing some files I had previously analysed in GGIR1.5-21 and came across an issue with valid days in version 1.5-24:

In 1.5-24 it appears to count all days as valid. It gets the wear_duration and 24 h cycle correct, it is just the valid days number. See attached, particularly person 8.
[part2_summary_vers21.xlsx](https://github.com/wadpac/GGIR/files/2189474/part2_summary_vers21.xlsx)
[part2_summary_ver24.xlsx](https://github.com/wadpac/GGIR/files/2189475/part2_summary_ver24.xlsx)

Let me know if it would be helpful to have any meta files

Thanks,
Alex

vincentvanhees, 2018-07-12:
thanks Alex, I will look into this

closed 2018-07-22 by vincentvanhees

vincentvanhees, 2018-07-22:
I have fixed it here on GitHub, and the changes will be incorporated in the next CRAN release of GGIR

commit 0a300f8f (2018-07-22): N valid days now correctly calculated in part2/g.analyse (was wrong after 1.5-21) closes #101, also extended the maximum measurement duration to 2 months, closes #101