wadpac/GGIR #105: LIDS analysis

vincentvanhees, 2018-08-09:
Expand GGIR with an implementation of the Winneback LIDS analysis.

**Outstanding action points:**
- [ ] Address Eva's comments on script (1/8/2018)
- [ ] Send Eva response re. those comments
- [ ] Send Eva example AX3 file
- [ ] Add functionality to export LIDS data, including: LIDS in normalised, non-normalised, fitted, cycle, period, etc.
- [ ] Wrap up as test version for others to play with
- [ ] Implement plateau searching
- [ ] Go once more through implementation and improve comments
- [ ] Add output variable on the duration of the wakefulness bout that succeeds the sleepbout.
- [ ] Improve terminology and variable names
- [ ] Improve the documentation and store an example R script somewhere online that stepwise shows how to run the LIDS analysis.
- [ ] Check that all key arguments can be manipulated with g.shell.GGIR and are not hardcoded
- [ ] Add max and median correlation to output

**Low priority:**
- [ ] Looping through multiple sleepbouts and add boutnumber
- [ ] Add option to truncate sleepbout length to 3 hours.
- [ ] Add option to change epoch resolution (currently at 1 minute).

EvaWinnebeck, 2018-08-13:
Hi Vincent,

did you have the following somewhere hidden on your to-do-list? 

1) output of ENMO (5sec) of the over3h-bouts for systematic testing of the conversion to LIDS, LIDS analyses, etc?

2) prior wakefulness: did this include the wakefulness since the last sleep as well as the wakefulness over the last 24h?

Cheers,

Eva

EvaWinnebeck, 2018-08-13:
Hi Vincent,

here is my first round of comments and suggestions on the current g.LIDS.analyse (https://github.com/wadpac/GGIR/blob/10bb7bba8280694f80ee1b0a03122d9dd58b1597/R/g.LIDS.analyse.R). I still need to go into your LIDS stationary analysis - but looks great already at first glance. Thank you!

- LIDS_S output PRIORITY: Please could you extend the LIDS_S output to the same breadth as the LIDS_NS output? i.e. add variables analogous to LIDS_NS's time, LIDSraw, LIDSfitted, LIDSfitted_abs, cycle, cycle_full, cycle_interpol, etc. This is really crucial as this will - at least initially - be our main basis for analysis. The output with LIDS_NS is - for now - only additional.

- Line 33 PRIORITY: threshold for binary 20 (in mg): Sorry, I just want to make absolutely sure the UKBiobank AX3 data are scaled in mg. My doubts are caused by the following: Although you have remarked that the AX3 files have ENMO in **mg** after GGIR processing, the cwa file (001_39434_0000000003_23-02-2018.cwa) that I received from you for tests definitely has ENMO in **g** after processing it with GGIR (for the cutoff to work, I either need to rescale ENMO by *1000 or adjust the threshold to 0.02). Was the test file not an AX3 file after all then or is the UKBiobank somehow processed differently? 

- Line 37: rollsum: We should add the option fill=NA here, so as not to drop the edges. This makes sure the entire original timeline is maintained and the later output in minutes reflects true "time since sleep onset" = biologically important.

- Line 45 PRIORITY: rollmean: The option fill=extend just creates a plateau at the edges over 15 min. This is basically data without meaning for LIDS and is likely to distort the analysis. Therefore, we should replace with fill=NA instead of to keep the original timeline – see above). 

- Line 37 and 45: edges rolling mean and sum: If we want to reduce the loss at the edges of the time series caused by the two filters, we should consider switching to zoo::rollapply with the option partial=TRUE. This uses partial filter sizes so that the data points created incorporate at least some of the neighbouring points. We used this in our paper (sorry, don’t think I made this explicit in the methods) for the 30’ rollmean. However, this was on a 10-min basis i.e. just affected the first data point which was then the mean of two instead of three data points. Rollapply is apparently not optimised for speed as much as rollsum or rollmean, however.

vincentvanhees, 2018-08-13:
Thanks Eva

1. At the moment I export the 5 second ENMO values by default for the whole measurement. Also exporting this per sleepbout would result in redundant data storage. Instead, I am exporting the 1 minute averages as part of LIDS_NS because those are at a 1 minute resolution. I understand that that is not really what you want. So, to solve this I could export the start and end time of the sleepbouts such that these time windows can be linked to the 5 second resolution ENMO values.

    - [ ] Export start and end times of sleep bouts

2. So far, I have only extracted the duration of the wakefulness period that directly follows the sleepbout. Wakefulness in the present night or in the 24 hours corresponding to the day on which the night started is stored elsewhere in the GGIR output. This is not compatible with the amount of wakefulness in the 24 hours preceding the sleepbout as you requested.

    - [ ] Investigate what we can do about this.

3. LIDS_S: I see what I can do. I think the first focus should be on having summary measures that are comparable such that we can start to get an idea of whether age group or sex differences can be reproduced.

    - [ ] Expand LIDS_S with time series

4. Units of acceleration are standardised in GGIR. After the data is first loaded it is stored in g, while in this part of the code they are multiplied by 1000 because the output needs to be in mg. One of the key characteristics of GGIR is that it takes care of this. So, no action needed.

5. Ok, thanks:

    - [ ] Implement fill=extend and partial=true.

EvaWinnebeck, 2018-08-13:
Hi Vincent,

here is my second round of comments and suggestions on the current g.LIDS.analyse (https://github.com/wadpac/GGIR/blob/10bb7bba8280694f80ee1b0a03122d9dd58b1597/R/g.LIDS.analyse.R).

- LIDS stationary fit: looks great! It produces slightly different results than my code in the RoO, cor and thus MRI output, but the general pattern is highly similar to my output and thus the selected period is the same - at least in the one night that I tested it in so far.

- LIDS_NS output: when I run the full function or just the lines used to create the NS output, all results pertaining to the cos fit are NA. PCNS remains empty (NA) as well. I have not yet trialled further where the problem originates.

-

EvaWinnebeck, 2018-08-13:
Hi Vincent, 

my answers to your new action points above (don't seem to be able to find a good way to quote your issues, sorry, I am sure there is one!):

1) ENMO of sleep bouts:
Great idea! Going from start and end points should work well in principle - if one has the full GGIR output. But will we be able to get this from Exeter?

2) Wakefulness - no immediate priority: we need PRIOR wakefulness not FOLLOWING wakefulness. This is important to judge sleep homeostasis. PRIOR wakefulness till previous bout and PRIOR wakefulness in the preceeding 24hours. But this is not essential for the first test analysis.

3. LIDS_S output - HIGH PRIORITY:
Several of the stats we do relies on the time series itself, LIDSraw, cycles, cycles_norm, all plots rely on the normalised dataset.

4) ENMO units: great!

5) Edges of filters:
Please don't implement fill=extend but either fill=NA or partial=TRUE.

vincentvanhees, 2019-02-16:
Update: Latest version of the code is in this branch: https://github.com/wadpac/GGIR/tree/LIDSModificationsEva
This is now up to date with latest version of the GGIR master branch (1.8-1).

In addition to the action points listed above I noticed that:
- [ ] variable "LIDS_sleepboutstart_char" in the part5_daysummary_WW_...csv output has value 1, probably this should be a character with a timestamp. It is not key to the LIDS analysis, but probably good to get this fixed.

- [ ] functions g.LIDS.analyse_ENMOsub and g.LIDS.analyse are very similar. Address redundancy by adapting function g.LIDS.analyse such that it can cover the functionality of both functions. 

- [ ] the time series related to the LIDS analysis have a lot of constants, would be good if we can avoid this unnecessary data storage, e.g. store constants in separate file like the part5_daysummary file which already has one row per night.

vincentvanhees, 2019-07-09:
I have now merged the g.LIDS.analyse and g.LIDS.analyse_ENMOsub function to ease future development and maintenance.

Also, I have added the average acceleration and activity score per minute to the time series that are store to a csv file. Note that these cannot be used to recalculate the LIDS scores, because LIDS scores are derived from the 5 second resolution data and only aggregated per minute afterward.

Further, I have merged in the latest version of the master branch such that this branch is up to date again with other developments in GGIR.

EvaWinnebeck, 2019-07-10:
Hi Vincent, 

Thanks for the update!

Please ignore all action points from last year as they were about a very different version of the code.

Re. current action points:

> I have now merged the g.LIDS.analyse and g.LIDS.analyse_ENMOsub function to ease future development and maintenance._
>
Sounds good if you prefer it that way. I had kept the two functions completely separate on purpose , so that we can simply ditch one once we know which one is better. I don't think we need both options long-term.

> Also, I have added the average acceleration and activity score per minute to the time series to the csv output file. Note that these cannot be used to recalculate the LIDS scores, because LIDS scores are derived from the 5 second resolution data and only aggregated per minute afterward.
> 
Great, thanks!

> Further, I have merged in the latest version of the master branch such that this branch is up to date again with other developments in GGIR.
>
Great, thanks!

Further action points discussed on your visit to Munich:
- [ ] How can we output the ENMO (5sec) of all sleep bouts for better LIDS development?

- [ ] Extend analysis to also include other bouts of each SPT/night (not only bout1)

- [ ] LIDS visualisation

- [ ] Add sleep homeostasis variables: 1) prior wake (between last bout and this bout); 2) sleep/wake history over prior 24h - let's discuss via skype how to best estimate both based on your sleep detection

Thanks, Eva

vincentvanhees, 2019-07-11:
Hi Eva,
The ENMO data is available, but possibly not in the most easy way. The 5 second ENMO values for the full recording are stored in the meta/ms2.out output folder in the IMP$metashort object. To extract the ENMO values corresponding to each 'sleep bout' used in the LIDS analysis the user will need to look up the variable _LIDS_sleepboutstart_num_ in the day-level part 5 output (csv file stored in output folder 'results').  This variable is the numeric time relative relative to the previous midnight expressed in hours on which the 'sleep bout' started. Variable _LIDS_sleepboutstart_char_ next to it gives the same time in clocktime for extra clarity. By combining the date and time of in the part 5 output, it will be possible to retrieve the corresponding ENMO values.

Variable _LIDS_sleepboutstart_char_ errornesly resulted in value 1, which I fixed today.

Other action points sound good. It would be good if we (or somebody else) can find dedicated funding to do them as they easily take up a month of work.

Best, Vincent

PS. I understand "I had kept the two functions completely separate on purpose" , but this is a nightmare to work with, because every time I fix a bug I need to remember to fix it in both versions of the code. It is easier to add it as a subroutine, which can be removed at a later point.

closed 2022-02-01 by vincentvanhees