wadpac/GGIR #108: Error in Math.factor(1L) : ‘round’ not meaningful for factors

KHC745, 2018-08-21:
Hi all,
I am trying to run GGIR with sleep log data and apparently it stopped at Part 4 where this error popping up:Error in Math.factor(1L) : ‘round’ not meaningful for factors.
I guess it has something to do with the sleep log format; however, I have prepared the sleep log in csv format as per recommended in the manual and thus I am not too sure what it meant. Hopefully someone can tell me which part went wrong! Thanks!

> g.shell.GGIR(mode=c(1,2,3,4), datadir="/Users/khc745/Documents/GENEActiv analysis/Data", outputdir="/Users/khc745/Documents/GENEActiv analysis/Results", f0=1, f1=1, do.enmo = TRUE, do.anglez=TRUE, chunksize=1, printsummary=TRUE, strategy = 2, ndayswindow=6, hrs.del.start = 0,hrs.del.end = 0, maxdur = 0,includedaycrit = 10, winhr = c(5,10), qlevels = c(c(1380/1440),c(1410/1440)), qwindow=c(0,24), ilevels = c(seq(0,400,by=50),8000), timethreshold=c(5), ignorenonwear=TRUE, excludefirstlast = FALSE, includenightcrit = 16, loglocation="/Users/khc745/Documents/GENEActiv_sleeplog/sleeplog.csv", outliers.only = TRUE, criterror = 4, relyonsleeplog = TRUE, sleeplogidnum = TRUE, colid=1, coln1=2, do.visual = TRUE, nnights =6, do.report=c(2,3,4), visualreport=TRUE, dofirstpage = TRUE, viewingwindow=1)

_____________________________________________________________________________________________________________________________________________________________
Part 1

_____________________________________________________________________________________________________________________________________________________________
Part 2

_____________________________________________________________________________________________________________________________________________________________
Part 3

_____________________________________________________________________________________________________________________________________________________________
Part 4
Error in Math.factor(1L) : ‘round’ not meaningful for factors

[image]

vincentvanhees, 2018-08-23:
you probably need to use sleeplogidnum = FALSE because your identifiers seems to be in character form. 
Thanks for formulating your question so clearly!

KHC745, 2018-08-26:
Hi Vincent,

Thank you for your quick response on this matter. I tried to use sleeplogidnum = FALSE but then there's another error coming up which I have also seen it before - not sure what it meant...
[image]

KHC745, 2018-08-28:
Hi Vincent,

Thank you for your prompt response. I had a look at the summary report for Part 2 and the id format is similar to the one I put in sleeplog (e.g. A02). 
[image]

I tried to play around with the argument idloc (= 1 or 2) too but the error is still there:
[image]
[image]

It seems to me that it can't read the sleeplog file - but I am not sure which part went wrong as I did it as per the documentation.

vincentvanhees, 2018-09-04:
Possibly something is wrong with the formatting of the csv file, e.g. tab rather than comma separated values. Try to assess how the values are separated, e.g. by opening the file in a text editor like notepad.

To speed up your investigation, you may want to first investigate whether function g.loadlog in GGIR can load your data. This function is responsible for loading the sleeplog file. See the function documentation for further instructions.

KHC745, 2018-09-05:
Hi Vincent,

Thanks for the suggestion. I have opened the csv file using notepad and the format seems correct i.e. values are separated by comma.
[image]

However, the same error occurred when I tried to load the sleep log file using the g.loadlog function. 
[image]

I wonder if there could be other errors for that to happen?

KHC745, 2018-09-05:
Hi Vincent,
Sure. Attached the sample of the sleeplog for your perusal. 

(ps. I can't upload the csv file over here so i saved it as XLSX. You may need to convert the file into CSV format.)
[sleeplog_trial.xlsx](https://github.com/wadpac/GGIR/files/2354824/sleeplog_trial.xlsx)

vincentvanhees, 2018-09-23:
You do not have a waking up time for the 7th night. The function expects two columns per night, one for going to bed and one for waking up. I understand that you may not have that data, so in that case you should still create the 15th column, but just leave it empty. What matters is that there are 15 columns (14 for the timestamps and 1 for the person identifier).

closed 2018-09-23 by vincentvanhees