wadpac/GGIR #113: Detection of sleep period without using sleep log data

KHC745, 2018-09-13:
Hi again,

I have had difficulties loading the sleep log dataset into R and thus I am trying to analyse the data without using the sleep logs. However, the results seemed strange as the sleep period was detected to be during daytime (in the afternoon) for all measurement days while my participants were all schoolchildren. I am not sure if I have made any mistakes in the code writing:

g.shell.GGIR(mode=c(1,2,3,4), datadir="/Users/khc745/Documents/GENEActiv analysis/Data", outputdir="/Users/khc745/Documents/GENEActiv analysis/Results", f0=1, f1=1, do.enmo = TRUE, do.anglez=TRUE, chunksize=1, printsummary=TRUE, strategy = 2, ndayswindow=6, hrs.del.start = 0, hrs.del.end = 0, maxdur = 7,includedaycrit = 10, winhr = 5, qwindow=c(0,24), timethreshold=5, anglethreshold=5, ignorenonwear=TRUE, excludefirstlast = FALSE, def.noc.sleep=c (19,8), do.visual = FALSE, nnights =6, do.report=c(2,3,4), visualreport=TRUE, dofirstpage = TRUE, viewingwindow=1)

[image]

Thanks in advance for the advice!

KHC745, 2018-09-19:
Hi, I've found the solution - it is due to the time zone difference.

closed 2018-09-19 by KHC745

lydiantoinette, 2018-10-31:
Hi Vincent, It seems like I'm still getting the same issue after putting in the desiredtz argument.
I've copied & pasted my code below - am I inputting this argument into the right parts?

## Not run:
mode= c(1,2,3,4,5)
datadir= "008.bin"
outputdir= "results"
studyname="RelapsePrediction"
f0 = 1
f1 = 2
g.shell.GGIR(#-------------------------------
             # General parameters
             #-------------------------------
             mode=mode,
             datadir=datadir,
             outputdir=outputdir,
             studyname=studyname,
             f0=f0,
             f1=f1,
             overwrite = FALSE,
             do.imp=TRUE,
             idloc=1,
             print.filename=FALSE,
             storefolderstructure = FALSE,
             #-------------------------------

             # Part 1 parameters:
             #-------------------------------
             windowsizes = c(5,900,3600),
             do.cal=TRUE,
             do.enmo = TRUE,
             do.anglez=TRUE,
             chunksize=1,
             printsummary=TRUE,
             derisedtz= "America/Toronto",
             #-------------------------------
             # Part 2 parameters:
             #-------------------------------
             strategy = 1,
             ndayswindow=7,
             hrs.del.start = 1,
             hrs.del.end = 1,
             maxdur = 9,
             includedaycrit = 16,
             L5M5window = c(0,24),
             M5L5res = 10,
             winhr = c(5,10),
             qlevels = c(c(1380/1440),c(1410/1440)),
             qwindow=c(0,24),
             ilevels = c(seq(0,400,by=50),8000),
             mvpathreshold =c(100,120),
             #-------------------------------
             # Part 3 parameters:
             #-------------------------------
             timethreshold= c(5,10),
             anglethreshold=5,
             ignorenonwear = TRUE,
             #-------------------------------
             # Part 4 parameters:
             #-------------------------------
             excludefirstlast = FALSE,
             includenightcrit = 16,
             def.noc.sleep = c(),
             loglocation= c(),
             outliers.only = FALSE,
             criterror = 4,
             relyonsleeplog = FALSE,
             sleeplogidnum = TRUE,
             colid=1,
             coln1=2,
             do.visual = TRUE,
             nnights = 9,
             #-------------------------------
             # Part 5 parameters:
             #-------------------------------
             # Key functions: Merging physical activity with sleep analyses
             threshold.lig = c(30,40,50),
             threshold.mod = c(100,120),
             threshold.vig = c(400,500),
             excludefirstlast = FALSE,
             boutcriter = 0.8,
             boutcriter.in = 0.9,
             boutcriter.lig = 0.8,
             boutcriter.mvpa = 0.8,
             boutdur.in = c(10,20,30),
             boutdur.lig = c(1,5,10),
             boutdur.mvpa = c(1,5,10),
             timewindow = c("WW"),
             #-----------------------------------
             # Report generation
             #-------------------------------
             do.report=c(2,4,5),
             visualreport=TRUE, dofirstpage=TRUE,
             viewingwindow = 1)
## End(Not run)

Thanks,
Lydia

vincentvanhees, 2018-11-04:
One possible issue with you code is that you have overwrite = FALSE, 
With this setting GGIR will not try to reprocess the files you already processed and you will continue to see the same problem. Change it to overwrite = TRUE, and run all parts again.

lydiantoinette, 2018-11-05:
Thank you Vincent, I tried that change but unfortunately am still seeing the ~ +4 hour time shift. Let me know if you have any other ideas of what could be wrong. I'm not sure if this will help, but have included a short personal test data file here. 
[image]

vincentvanhees, 2018-11-19:
Did you process part 1 again? This is the part where the interpretation of timezone primarily takes place.