wadpac/GGIR #119: Sleep_efficiency

albacabre, 2018-09-28:
Dear all,

I am trying to figure out how sleep_efficiency is calculated by GGIR. As far as I understood reading the literature it supposed to come out from acc_SleepDurationinSPT (or acc_dur_noc in older versions) / acc_sptduration (acc_timeinbed in older versions). However, I double checked it manually and it gave me different values (e.g. mean sleep efficiency calculated with GGIR is 0.818 and manually calculated is 0.84.)

Moreover, some individuals have information on acc_SleepDurationinSPT  and  acc_sptduration but sleep_efficiency is missing.

Does anyone have a clue what could be the problem? 

Many thanks,

vincentvanhees, 2018-10-06:
The sleep efficiency  is calculated per night and then averaged across nights. The order of the calculation is critical when you try to reproduce it. If you first calculate the average acc_SleepDurationinSPT and average acc_spitduration across nights and then take the ratio you are likely to arrive at different numbers.

The summaries per measurement are derived by aggregating the estimates per night, so if you want to check the calculation it is best to focus first on the night specific estimates.

- Are the values missing in the night specific reports (part4_nightsummary_sleep_cleaned.csv) or in the recording summary (part4_personsummary_sleep_cleaned.csv)?
- Do the missing values coincide with nights for which there is large proportion of invalid data (e.g. monitor not worn)?

closed 2018-10-28 by vincentvanhees

albacabre, 2018-11-09:
Hi Vincent,

Thanks for your feedback, I have been checking the data since then and still cannot see why if for subject 1 we have recorded 8 nights at the end we are only getting sleep efficiency for two nights. I have checked what you suggested but I think these night we loose do not have much invalid data.

In this [shared ](https://drive.google.com/drive/folders/1mR2vOg2o7700es_u45h_Fmnq2458dNnw?usp=sharing)folder there is an excel file with an example using one participant (part2, part4, and part5 are all summarized in one excel sheet). Can you see why part5 only gives us information for two night?

Many thanks!