wadpac/GGIR #123: Can GGIR run Axivity files converted to .csv?

Drew74124, 2018-10-16:
Hi, 
We have axivity files converted to .csv, and I can't seem to get them to run through GGIR.

The format is shown below. First column is a time stamp and the following are x,y,z.

00:04.5 | 0.171875 | 0.9375 | -0.14063
00:04.5 | 0.1875 | 0.984375 | -0.15625
00:04.5 | 0.171875 | 0.984375 | -0.15625
00:04.6 | 0.1875 | 0.984375 | -0.15625
00:04.6 | 0.1875 | 0.984375 | -0.14063
00:04.6 | 0.1875 | 0.984375 | -0.15625
00:04.6 | 0.1875 | 0.984375 | -0.15625

Is it possible to run these? or convert them to something that can run?

Cheers
Drew

vincentvanhees, 2018-10-28:
Not yet. Note to other readers: Drew and I are looking into this off-line.

vincentvanhees, 2018-11-25:
I have merged the work with the master branch to be included in the next GGIR release. I will close this issue now, but let me know if you encounter any issues.

closed 2018-11-25 by vincentvanhees