wadpac/GGIR #132: Ready to process Axivity data

gradidgep, 2018-11-12:
Cannot process Axivity data
Hi there
Im new to the Axivity. Trying to process data in R studio but get the following error message. Can someone please help;

Part 1
[1] "P1 file1"
[1] "---------------------------------------------"
[1] "investigate calibration of the sensors..."

Loading block: 1 2
End of file reached

stop reading because there is not enough data in this block

----------------------------------------
Summary of autocalibration procedure:

Status:
[1] "recalibration attempted with all available data, but possibly not good enough: Check calibration error variable to varify this"

Calibration error (g) before:
[1] 0.06859568

Callibration error (g) after:
[1] 0.002175918

Offset correction:
[1] 0.02531877 0.00200245 0.12766943

Scale correction:
[1] 0.9984907 1.0051335 1.0297449

Number of hours used:
[1] 12

Number of 10 second windows around the sphere:
[1] 312

Temperature used (if available):
[1] TRUE

Temperature offset (if temperature is available):
[1] -0.003258179  0.001685339 -0.009777969

----------------------------------------
[1] "get meta data..."

Loading chunk: 1
End of file reached
[1] "save .RData-file with: calibration report, file inspection report and all meta data..."

______________________________________________________________________________________________
Part 2
 1
______________________________________________________________________________________________
Part 3
 1
______________________________________________________________________________________________
Part 4
 1Error in readChar(con, 5L, useBytes = TRUE) : cannot open the connection
In addition: Warning message:
In readChar(con, 5L, useBytes = TRUE) :
  cannot open compressed file 'C:/Users/a0016736/Documents/Axivity/studentj/output_studentj/meta/ms3.out/NA', probable reason 'No such file or directory'

gradidgep, 2018-11-26:
> Use visualreport=FALSE if you have not used mode = 3, 4 and 5 yet.
> Do not use do.report = c(4) when you have not used mode = 3 and 4 yet.

Hi Vincent
I have run the script again with your suggested edits.
I do not get a csv-file written by GGIR. In fact, I never have. I always
get 2 folders, one called "meta", the other called "results".
the Results folder has a QC folder with 3 files, filelocation.csv,
sessioninfo_part1.Rdata, and sessioninfo_part2.Rdata. the meta folder has a
ms2.out folder and a folder called basic. the ms2.out folder is empty, and
the basic folder has a file called meta_35406_0000000000.cwa.Rdata
is there perhaps something additional to install on my side?
thanks

vincentvanhees, 2018-12-02:
Maybe there is not enough data in the file, because normally an output report should be created.
- Do you have a file "data_quality_report.csv" in your results/QC folder? This should indicate whether the file was considered corrupt or too short. 

Also, you may want to try:
- Use argument `strategy = 1` and `chunksize = 1`
- Reprocess the file (argument `overwrite = TRUE`) with the latest version of GGIR (1.7-1), because there was an issue with cwa file reading which should be fixed now. 

If you have < 24 hours of data a lot of the functionality will not be used. GGIR is designed for multi-day measurements.  Probably good to verify in your OMGUI software that you have at least a couple of days of data in your test files.
Best. Vincent

gradidgep, 2018-12-04:
Dear Vincent
Thank you for your suggestion, but it still doesnt work unfortunately. I
checked the file in the OMGUI. There is >24 hours of data. In most cases
the participants are wearing for the AX3 monitors for 7 days.
In addition, I dont get a "data_quality_report.csv" file in the QC folder.
Regards
Philippe

vincentvanhees, 2019-01-21:
The updates in the latest GGIR version (1.8-1) address should have addressed the problem.
If not, then please re-open this issue. Thanks, Vincent

closed 2019-01-21 by vincentvanhees