wadpac/GGIR #133: g.cwaread sometimes fails to read entire file

vincentvanhees, 2018-11-17:
While investigating [this](https://groups.google.com/forum/#!topic/rpackageggir/20Raobr0Cws) issue from the GGIR google group it became clear that g.cwaread sometimes fails to read the entire cwa file. The only way to notice this is by comparing the visual signal in OMGUI with the result in GGIR. The most likely explanation is data corruption. I am now working on a solution and will reference this issue when committing changes.

vincentvanhees, 2018-11-17:
GGIR reads the data in blocks of about 24 hours. Each block in turn is made up of data pages, where each page is 6 seconds. When the function g.cwaread failed to read one of those blocks, it actually stumbles on only the first page.  I have now added a routine to skip the first page, try whether the second page loads (if not, then continue with 3rd, 4th, ....10th page). Once a page can be loaded it will consider that page to be the beginning of the block. Next, the block is loaded at the first page of the block is repeated to compensate for the pages that were skipped. This seems to work as I am able to read the problematic files now. I have saved the changes in the cwaread_issue133 branch of the repository.

Although the above seems to work, I do not understand why those pages were unreadable, so this requires further investigation.

vincentvanhees, 2018-11-25:
I am closing this issue now the fix to skip a file page has been merged.

closed 2018-11-25 by vincentvanhees

vincentvanhees, 2019-01-09:
Update: Evgeny has been able to fix the underlying cause of the problem which was in the g.cwaread code. I have now merged this into the master branch and will prepare a new GGIR release later this week.

muschellij2, 2020-04-08:
Getting `Error in if (x >= 512) { : missing value where TRUE/FALSE needed` from a UK Biobank AX3:
```
res = GGIR::g.cwaread("1234520_90001_0_0.cwa", start = 0, end = 1e6)
```

Failing at `49%`.  Can't share the data 

Traceback:
```
4: numUnpack2(packedData)
3: readDataBlock(fid)
2: GGIR::g.cwaread(fileName = file, start = 0, end = end, progressBar = verbose, 
       ...)
```

vincentvanhees, 2020-04-09:
Can you re-try with the master branch on GitHub. 

- Function numUnpack was not able to handle accelerometer recordings with dynamic ranges other than +/-8g.
- To address this function numUnpack2 was created earlier this year with help of Dan Jackson, which fixed the issue but was too slow as it was written in R.
- Most recently I used that R code to correct the original numUnpack c++ function and removed numunpack2, but forgot to add numUnpack back to the NAMESPACE file, which I corrected today.
- So, this should hopefully address the issue and I will include this in the next CRAN release. The latest CRAN release is from early December and should not have the issue you are referring to as it still works with the old numUnpack function.

muschellij2, 2020-04-09:
Works great, thanks