wadpac/GGIR #144: Cannot process Actigraph GT3X data

Anjolu, 2018-12-03:
Hi there,

I ran into a problem when trying to run the data in R studio. Hope anyone could help. The error message is as below:

"Loading block: 1Error in data.table::fread(filename, nrow = 2, skip = 10, dec = ",", showProgress = FALSE,  : 
  Unable to change to a locale which provides the desired dec. You will need to add a valid locale name to getOption("datatable.fread.dec.locale"). See the long paragraph in ?fread. Run again with verbose=TRUE to inspect."

(ps. When I tried the to use function fread from the data.table package directly on my csv files and changed the dec value as ".", the CSV. file finally got recognised. So I was wondering what should I do?)

I also attached an example file for test. 
Thanks so much!

Anjo

[hk0002 (2016-05-10)RAW.zip](https://github.com/wadpac/GGIR/files/2639959/hk0002.2016-05-10.RAW.zip)

vincentvanhees, 2018-12-03:
This should be fixed now here in the GitHub code following [this commit](https://github.com/wadpac/GGIR/commit/5b75015699fc4e0cdd5b5a899f095014f04341ca), and I will include this in the next release of GGIR. Thanks

closed 2018-12-03 by vincentvanhees

Anjolu, 2018-12-07:
> This should be fixed now here in the GitHub code following [this commit](https://github.com/wadpac/GGIR/commit/5b75015699fc4e0cdd5b5a899f095014f04341ca), and I will include this in the next release of GGIR. Thanks

Dear Vincent, 

Thanks for the reply. But I am still confused about how I should change the codes in the g.shell.GGIR? Could you please advice? Thanks!

Anjo

vincentvanhees, 2018-12-10:
This is a change to GGIR itself, not to the GGIR function calls.
If you are working directly with the code in this repository, it should work.
If you are working with the GGIR version on CRAN, then wait for the next version.

Anjolu, 2018-12-14:
> This is a change to GGIR itself, not to the GGIR function calls.
> If you are working directly with the code in this repository, it should work.
> If you are working with the GGIR version on CRAN, then wait for the next version.

Thanks Vincent! I would try that codes.

Anjo

Anjolu, 2018-12-14:
> This is a change to GGIR itself, not to the GGIR function calls.
> If you are working directly with the code in this repository, it should work.
> If you are working with the GGIR version on CRAN, then wait for the next version.

Thanks Vincent! I will try that code.

Anjo

commit d590d0e3 (2018-12-03): updated changelog. fixes #144