wadpac/GGIR #145: Error in g.part5(datadir = datadir, metadatadir = metadatadir, f0 = f0,  :    object 'qqq2' not found

lydiantoinette, 2018-12-03:
Hello,
I seem to be getting an error in part5:
Error in g.part5(datadir = datadir, metadatadir = metadatadir, f0 = f0, : object 'qqq2' not found. 

I've been able to generate the reports in the past, but it seems to give me this issue after I've updated my "desiredtz" argument. I've restarted R Studio & re-installed GGIR as well. 

Below is my code: 
# NOT RUN {
mode= c(1,2,3,4,5)
datadir= "test.bin"
outputdir= "results"
desiredtz="US/Eastern"
studyname="DEW"
f0 = 0 
f1 = c()

g.shell.GGIR(#-------------------------------
             # General parameters
             #-------------------------------
             mode=mode, 
             datadir=datadir, 
             outputdir=outputdir, 
             studyname=studyname, 
             f0=f0, 
             f1=f1,
             desiredtz=desiredtz,
             overwrite = FALSE, 
             do.imp=TRUE,
             idloc=1, 
             print.filename=FALSE,
             storefolderstructure = FALSE,
             #-------------------------------
             # Part 1 parameters:
             #-------------------------------
             windowsizes = c(5,900,3600),
             do.cal=TRUE, 
             do.enmo = TRUE,
             do.anglez=TRUE,
             chunksize=1,
             printsummary=TRUE,
             desiredtz=desiredtz,
             #-------------------------------
             # Part 2 parameters:
             #-------------------------------
             strategy = 1,
             ndayswindow=7,
             hrs.del.start = 1,
             hrs.del.end = 1, 
             maxdur = 9,
             includedaycrit = 16, 
             L5M5window = c(0,24),
             M5L5res = 10,
             winhr = c(5,10),
             qlevels = c(c(1380/1440),c(1410/1440)),
             qwindow=c(0,24), 
             ilevels = c(seq(0,400,by=50),8000), 
             mvpathreshold =c(100,120),
             #-------------------------------
             # Part 3 parameters:
             #-------------------------------
             timethreshold= c(5,10),
             anglethreshold=5,
             ignorenonwear = TRUE,
             #-------------------------------
             # Part 4 parameters:
             #-------------------------------
             excludefirstlast = FALSE,
             includenightcrit = 16,
             def.noc.sleep = c(),
             loglocation= c(),
             outliers.only = FALSE,
             criterror = 4,
             relyonsleeplog = FALSE,
             sleeplogidnum = TRUE,
             colid=1, 
             coln1=2, 
             do.visual = TRUE,
             nnights = 9,
             #-------------------------------
             # Part 5 parameters:
             #-------------------------------
             # Key functions: Merging physical activity with sleep analyses
             threshold.lig = c(30,40,50),
             threshold.mod = c(100,120),
             threshold.vig = c(400,500),
             excludefirstlast = FALSE,
             boutcriter = 0.8,
             boutcriter.in = 0.9,
             boutcriter.lig = 0.8,
             boutcriter.mvpa = 0.8,
             boutdur.in = c(10,20,30),
             boutdur.lig = c(1,5,10),
             boutdur.mvpa = c(1,5,10),
             timewindow = c("WW"),
             #-----------------------------------
             # Report generation
             #-------------------------------
             do.report=c(2,4,5))
# }

vincentvanhees, 2018-12-03:
Interesting, why not stick with "America/Toronto" if that worked?

Are you sure that is the only thing you changed? According to https://en.wikipedia.org/wiki/List_of_tz_database_time_zones it should give the same result.

I agree that the error message "object 'qqq2' not found." is not very informative, but I mainly see it when users try to process different parts of GGIR with different desiredtz settings. To be sure you are not re-using data, delete the entire output folder and start again. A less rigorous option would be to set overwrite = TRUE and with mode = 1:5.

mltoledo, 2018-12-12:
We've had the same issue with our data. The problem seems to stem from this line in g.part5:

 if(length(unlist(strsplit(timebb[1],"[+]"))) < 2)

where "[+]" was specified to split the datetime. However, the datetimes in America have "[-]" (negative offset from UTC) which creates the "object 'qqq2' not found" error message.

vincentvanhees, 2018-12-12:
Thanks, that could indeed be the problem. I will try to work on a solution in the upcoming weeks.

lydiantoinette, 2018-12-14:
Thanks Vincent! 

(And I had tried with the overwrite = TRUE and modes = 1:5, but was getting the same error)

vincentvanhees, 2018-12-14:
I have made a pull request for the changes that should fix the problem, and asked Jairo to check and merge it, just as a double check that my solution is correct. Once this is merged we will include this in the next GGIR release on CRAN. Thanks again @mltoledo for pointing out the problem!

lydiantoinette, 2018-12-20:
Thanks very much @vincentvanhees and @mltoledo!

closed 2019-01-06 by vincentvanhees