wadpac/GGIR #149: GGIR results shifted one hour

jcapde, 2018-12-18:
Hi, 

We noticed that GGIR results for Axivity traces (in .cwa format) recorded in UK local time during British summer time are shifted one hour.  Results look good during winter period (GMT). We are currently using the default value for `desiredtz= "Europe/London"`. Any idea what might be going wrong?

Thanks in advance
Joan

vincentvanhees, 2018-12-18:
Thanks Joan, @Mirkes Do you have any initial thougths on this?

Summer time handling should be fine for all other data formats, but .cwa format reading functionality is a relatively new and not something I worked on a lot.

vincentvanhees, 2018-12-18:
Either way, I will try to have a look at it on Friday.

Mirkes, 2018-12-18:
Dear Vincent,

If I correctly remember that there is no information about time zone in the Axivity files. As a result I use standard time zone. It can be easily changed but I need to remember code and understand the way how to do this. I think that I can do this in week end.

Best regards,
Evgeny

vincentvanhees, 2018-12-18:
Thanks both.
Key processing steps in GGIR in relation to the timestamp are:
- timestamps are loaded via (g.cwaread -> [g.readaccfile](https://github.com/wadpac/GGIR/blob/master/R/g.readaccfile.R) -> [g.getmeta](https://github.com/wadpac/GGIR/blob/master/R/g.getmeta.R))
- interpretation of timestamp happens in  [g.getmeta](https://github.com/wadpac/GGIR/blob/master/R/g.getmeta.R)) -> [g.getstarttime](https://github.com/wadpac/GGIR/blob/master/R/g.getstarttime.R) where the code uses the input argument desiredtz to interpret the timestamp according to timezone. Which also applies to Axivity cwa format. (lines 5-9)

I have some files from summer and winter to investigate this on Friday and will reply to this issue with my findings.

My initial thought for now: It may be that for other data formats (incl. other accelerometer brands) the timestamps are stored by default in GMT and then interpreted based on desiredtz, while for Axivity cwa format the timestamps are already in DST after which a time conversion would be redundant. If so, then I hope that this is a consistent situation for all Axivity cwa data, and not dependent on how configure their data.

jcapde, 2018-12-18:
Thanks Vincent. I've changed our parameterisation to explicitly set `desiredtz= "Europe/London"` - we did not before since we've thought that that was the default value, and it seems to now generate the correct results. Hence, I think that the issue could be related to the default tz. I will do further testing and let you know.

Mirkes, 2018-12-18:
Dear Vincent,

I have reread g.cwaread.r and found that in this function I never used tz. I work with timestamp in two places:
In function timestampDecoder in the line  54:
      year = as.numeric(as.POSIXct(
        paste0(year, "-", month, "-", day, " ", hours, ":", mins, ":", secs)
      ))
In function readHeader I transform start time in the line 120
        start = as.POSIXct(datas$start, origin = "1970-01-01"),

Final transformation timestamps into string representation is produced outside of g.cwaread.r.

I hope it is useful.
Best regards,
Evgeny

vincentvanhees, 2019-01-04:
Thanks both. Argument desiredtz should indeed default to "Europe/London" if not provided by the user. This was the case for g.shell.GGIR and various other GGIR functions, but not for every single function inside GGIR, which seems to have explained the problem. Thanks for addressing this. I will now close this issue.

closed 2019-01-04 by vincentvanhees