wadpac/GGIR #162: improved documentation needed for sleep detection

vincentvanhees, 2019-01-22:
The sleep detection in GGIR was originally designed based on the assumption that a sleep log (diary) is available as described in our 2015 [paper](https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0142533).

However, in recent years GGIR has been expanded with several sleep detection options that do not rely on a sleeplog anymore. This makes that a lot of the terminology within GGIR has become confusing for those users without a sleeplog. In our recent [Scientific Reports paper](https://www.nature.com/articles/s41598-018-31266-z) I fully focused on describing the HDCZA algorithm as a replacement for the sleeplog, and only expanded the documentation to describe how to use this approach. As that paper already had a lot of content, I did not have space (or time) to also provide a detailed description of all the options associated with the sleep algorithms within GGIR and how their impact varies with the different SPT detection algorithms. The link to [the repository I used](https://github.com/wadpac/whitehall-acc-spt-detection-eval/blob/master) should hopefully clarify anything that the paper does not describe, but it is not ideal as a way to communicate how things work.

Proposed solution:
- Try to improve variable names (as also discussed [here](https://github.com/wadpac/GGIR/issues/160)).
- Try to improve function documentation and vignette.
- Provide clearer advise about what variables should be used: The acc_xxxx variables are prefered over the sleeplog_xxxx variables. They are both in the output to facilitate comparison, but for a new user it can be confusing.

JianWang2016, 2019-02-16:
@vincentvanhees 
It is great to provide alternative to sleeplog.  In the sentence "The link to [the repository I used](https://github.com/wadpac/whitehall-acc-spt-detection-eval/blob/master) should hopefully clarify...", the link is broken.
Thanks,

vincentvanhees, 2019-02-20:
thanks - url now fixed