wadpac/GGIR #168: Final report creation step is mixed with g.part1()

JianWang2016, 2019-02-20:
Hi Vincent:  I ran part 1 through part 5 sequentially on a cluster.  This last step is run to create reports.  I encountered a couple of issues here.

1) When overwrite=TRUE, g.part1() is accidentally activated; therefore I can only set overwrite=FALSE.  However, in order to rerun this step to create new reports, I need to manually remove the "file summary reports" folder to proceed.
2) At the end of this step, there is following error:
Error in xy.coords(x, y, xlabel, ylabel, log) : 
  'x' and 'y' lengths differ
3) viewingwindow can only be set to 1, midnight to midnight.  If I set it to 0 hoping to change to noon to noon, following error appears:
Generate visual reports
[1] "File filename.cwa"
Error in g.plot5(metadatadir = metadatadir, dofirstpage = dofirstpage,  : 
  object 'nightsi' not found

Following is the code for creating the reports. Please note all the previous steps are completed at this point.

Thanks so much for your help.

g.shell.GGIR(
  #=======================================
  
  #-------------------------------
  # General parameters
  #-------------------------------
  datadir = datadir,
  outputdir = outputdir,
  
  f0 = 1,
  f1 = c(),
  # file start and file end
  overwrite = FALSE,
  do.imp = FALSE,
  
  #-----------------------------------
  # Report generation
  #-------------------------------
  
  # Key functions: Generating reports based on meta-data
  do.report = c(2, 4, 5),
  visualreport = TRUE,
  dofirstpage = TRUE,
  viewingwindow = 1
  
) # end of g.shell.GGIR()

vincentvanhees, 2019-02-20:
1. Do not set mode  = 1 if you only want to regenerate the report. If mode = c() then part1 does not start. Note that reports are always created, even if overwrite = FALSE.
2. & 3. This is too little information for me. Can you narrow down when / where this happens? Does the file have at least 1 full day of measurement with the monitor being worn between noon-noon and midnight-midnight?

vincentvanhees, 2019-02-20:
Hi Jian,
The default for mode is mode=c(1,2)
If you do not want the part 1 and 2 to be processed again you will need to include mode=c() in your function call.
Best, Vincent

JianWang2016, 2019-03-14:
Vincent, 

When overwrite = TRUE, part 1 is still activated even if mode = c()

if overwrite = FALSE, part 1 is not activated even if don't set mode

JianWang2016, 2019-03-14:
when overwrite = FALSE (which is the only way to avoid trigging part 1 reran, mode = c() won't make a difference), some reports are created but reports within "file summary reports" folder will not be recreated unless this folder is manually removed first.

Yes, my samples have over 10 days of good data collection.

vincentvanhees, 2019-05-09:
Hi JIan,
We discussed this over e-mail, but for reference purposes for other users here a summary:

When making a repeated call to function g.shell.GGIR  make sure arguments `overwrite` and `mode` are always provided, because otherwise they use default values that may or may not align with what you want to achieve. It is better to just explicitly provide them.

Multiple visual reports are created by GGIR. Below I will focus on the visual report per measurement that has histograms on the first page. The following logic applies:

**Setting `mode =c( )`, `visualreport = TRUE`, and `overwrite = TRUE`** will create visual reports from the previously generated milestone files inside the 'meta' folder,  and overwrite the existing files inside the 'results/file summary reports' folder, if any.

**`Setting mode =c( )`, `visualreport = TRUE`, and `overwrite = FALSE`** will create visual reports from the previously generated milestone files inside the 'meta' folder, but only for files that do not have a report in the 'results/file summary reports' folder.

**Setting `mode =c(1, 2, 3, 4, 5)` or not providing the `mode` argument, `visualreport = TRUE`, and `overwrite = TRUE`** will re-run part 1-5, overwrite the corresponding milestone files in the 'meta' folder, and will create visual reports overwriting the existing reports inside the 'results/file summary reports' folder.

Note: Argument `do.visual` does not relate to the report discussed here, but relates to the horizontal bar plot produced in part 4 and stored in visualisation_sleep.pdf. If you want this file to be generated for a full dataset then run part 4 in series (non-parallel). It will go wrong if GGIR is applied in parallel, because than the parallel processes will overwrite the same file visualisation_sleep.pdf for each batch, unless you tell your parallel processing script to store each batch in a separate output folder.

Feel free to re-open this issue if you continue to experience problems with this.

closed 2019-05-09 by vincentvanhees