wadpac/GGIR #170: part 5 day summary adding zeros for some variables

Alex-Row, 2019-03-07:
In trying to marry up some results from part 5 (wake to wake) 
[part5_personsummary_WW_L40M100V400_sleep.T5A5.xlsx](https://github.com/wadpac/GGIR/files/2942460/part5_personsummary_WW_L40M100V400_sleep.T5A5.xlsx)
[part5_daysummary_WW_L40M100V400_sleep.T5A5.xlsx](https://github.com/wadpac/GGIR/files/2942461/part5_daysummary_WW_L40M100V400_sleep.T5A5.xlsx)

to part 4 I got some big differences in sleep duration. 

It seems that there is a zero included in the day summary files where an invalid day has been identified, e.g. see column P night 7 in attached day summary file. In the person summary this then gets included in the average bringing it down. Once I exclude that value it matches the values from part 4. 

I’ve only sent one person in the attached, but it seems to be happening across the whole sample.

Let me know if you need any more information.

vincentvanhees, 2019-03-10:
Thanks for raising this Alex.

This is because a night of length zero is included in the day summary file. That in itself is correct, because  last year I was told that g.part5 output should also show half WW (waking up to waking up) days at the beginning and end of the measurement.

Where it goes wrong is the person level aggregation. The person level aggregates are calculated by excluding the days with too much monitor wear time (not too little non-wear time), including any length of a day or night. This is where it goes wrong, days with night or daytime = 0 are included in the person level average. 

I have now updated the code for calculating the person level estimates to also exclude days when the daytime or nighttime has length zero. I imagine that some GGIR users are interested in getting an average per person that includes the daytime from that last day, but not the night time because it is zero. However, that is not what the code currently does. Possibly those estimates can be obtained from the MM window variant on the part 5 output, because there the day is aligned more favorable for missing first daytime and last nighttime.

It would be quite a lot of work to modify this such that the data is aggregated per variable type. The safest approach for now is probably to exclude all incomplete days for the personal level aggregates (as I have now done). The day summary output will remain unchanged. So, if desired you or other users can continue to explore alternative ways of aggregating the data from day level to person level.

I will include these updates in the next release.

Alex-Row, 2019-03-10:
Great, thanks Vincent.

PR #176 (2019-03-14): Issue170 part5mean
Fixes #170. person level aggregation by g.part5 now excludes days with day or night time duration of zero, because including them would not be meaningful.
closed 2019-03-14 by vincentvanhees