wadpac/GGIR #186: arm angle confusion?

Tim-Yu, 2019-05-12:
Hi, I am currently trying to regenerate 3d the walking pattern based on the AX3 data. But stuck at getting the correct arm angle.
In your metric.R, I found the your angle solution. 
```   
 if (ii == 11) { # angles
      anglex = (atan(Gxm / (sqrt(Gym^2 + Gzm^2)))) / (pi/180)
      angley = (atan(Gym / (sqrt(Gxm^2 + Gzm^2)))) / (pi/180)
      anglez = (atan(Gzm / (sqrt(Gxm^2 + Gym^2)))) / (pi/180)
      Gfil = cbind(anglex,angley,anglez)
    }
```
It seems that the angle is calculated based on the 'arm axes'.
[image]
But the axes are not fixed which may change when the arm rotates, moves, therefore in the final report,  what does the 'arm angle (up or down)' mean? 

Thanks in advance.

vincentvanhees, 2019-05-13:
Thanks Tim-Yu.
The term 'up or down' is applicable to any of the three angles x, y, and z, because each angle is the angle between a sensor axis and the horizontal plane. The direction of up or down will vary, which is why I have not added the direction of up and down to the axis label.

Axes are always local axes on the device, independent of the orientation of the wrist/arm, and never intended to be global axes.

The z-axis is always the axis perpendicular to the skin, no matter how you insert the Axivity sensor cube into the Axivity wrist strap. The only aspect that can change between experiments is the direction of positive and negative. This is why I choose to do the sleep classification only with the z-axis.

The term 'arm' in the axis label assumes that the sensor was worn on the wrist, 'arm' could be replaced by 'lower arm' or 'wrist'. Calling it 'estimate of sensor z-axis angle relative to the horizontal' would be more correct, but that would also make the visualisations too technical for users in the health/clinical domain.

If you have suggestions for improving the plots and/or documentation then let me know.
Best wishes, Vincent

vincentvanhees, 2019-05-13:
After reflecting more on this, I now realise that the mistake and your confusion is in the combined reference to arm and up/down orientation. How about we update 'arm angle (up or down)' to:

"z-axis angle relative to horizontal plane (for wrist attachment this equals hand palm orientation)".
This would be technically correct and interpret-able for a non-technician, agree?

Tim-Yu, 2019-05-13:
Hi, Vincent,
Thanks a lot for your explanation. I now understand the arm angle. I think 'arm angle' is more concise on the plot, but it will be great if adding some annotations at the end of the report explaining what is the arm angle referring to.

Best wishes,
Tim

vincentvanhees, 2019-07-03:
I have now addressed this and will include the change in the new GGIR version that I will upload to CRAN later today

closed 2019-07-03 by vincentvanhees