wadpac/GGIR #192: Problems running part5

katisarafrazi, 2019-06-19:
I am running GGIR on 9 hrs of data, and I am looking for a sleep/wake classification. But whwn its comes to part5 i get the error: " Error in `[.data.frame`(daysummary_tmp, , MainMetric) :  object 'MainMetric' not found"
Can anyone help me?

Below is top rows of a sample input csv file:
[image]

This is how i called the shell function:
g.shell.GGIR(mode=c(1,2,3,4,5),
             datadir,
             outputdir,
             overwrite = TRUE,
             strategy = 1,
             hrs.del.start = 0,          hrs.del.end = 0,
             maxdur = 2,                 
             qwindow=c(0,24),
             mvpathreshold =c(100),
             bout.metric = 4,
             excludefirstlast = FALSE,
             includedaycrit = 16,
             includenightcrit = 16,
             def.noc.sleep = c(1),
             outliers.only = TRUE,
             criterror = 4,
             do.visual = TRUE,
             threshold.lig = c(30), threshold.mod = c(100),  threshold.vig = c(400),
             boutcriter = 0.8,      boutcriter.in = 0.9,     boutcriter.lig = 0.8,
             boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig = c(1,10),
             boutdur.mvpa = c(1),   timewindow = c("WW"),
             do.report=c(2,4,5),
             visualreport=TRUE)

vincentvanhees, 2019-07-03:
The GGIR code in function g.plot5.R looks for a variable in the part2 daysummary report  that looks like: "mean_ENMO_mg_24h" or "mean_ENMO_mg_0-24hr". If this variable is not present then it cannot create the first page of the report.

Solution:
- Set argument dofirstpage =FALSE which should skip this part of the code.
or
- Only use files with at least one day of data, because then those variables will be created.

closed 2019-07-03 by vincentvanhees

katisarafrazi, 2019-07-08:
> The GGIR code in function g.plot5.R looks for a variable in the part2 daysummary report that looks like: "mean_ENMO_mg_24h" or "mean_ENMO_mg_0-24hr". If this variable is not present then it cannot [...]

Thank you!

AkhtarNishat, 2020-09-18:
> I am running GGIR on 9 hrs of data, and I am looking for a sleep/wake classification. But whwn its comes to part5 i get the error: " Error in `[.data.frame`(daysummary_tmp, , MainMetric) : object 'MainMetric' not found"
> Can anyone help me?
> Also are there any sample input files available?
> 
> Below is top rows of a sample input csv file:
> [image]
> 
> This is how i called the shell function:
> g.shell.GGIR(mode=c(1,2,3,4,5),
> datadir,
> outputdir,
> overwrite = TRUE,
> strategy = 1,
> hrs.del.start = 0, hrs.del.end = 0,
> maxdur = 2,
> qwindow=c(0,24),
> mvpathreshold =c(100),
> bout.metric = 4,
> excludefirstlast = FALSE,
> includedaycrit = 16,
> includenightcrit = 16,
> def.noc.sleep = c(1),
> outliers.only = TRUE,
> criterror = 4,
> do.visual = TRUE,
> threshold.lig = c(30), threshold.mod = c(100), threshold.vig = c(400),
> boutcriter = 0.8, boutcriter.in = 0.9, boutcriter.lig = 0.8,
> boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig = c(1,10),
> boutdur.mvpa = c(1), timewindow = c("WW"),
> do.report=c(2,4,5),
> visualreport=TRUE)

I am having the same problem. Can you help?

vincentvanhees, 2020-10-08:
Please provide a more detailed description with minimum reproducible example:
- RData file from the meta/basic folder
- R script
- Copy of error messages you get

vincentvanhees, 2020-10-08:
See also my message in the google group. I can prioritize this if paid as consultancy.

AkhtarNishat, 2020-10-13:
> Please provide a more detailed description with minimum reproducible example:
> 
> * RData file from the meta/basic folder
> * R script
> * Copy of error messages you get

AkhtarNishat, 2020-10-13:
I am a student. Using GGIR for analysing sensor data for breast cancer study. I can't pay much but I am willing to pay whatever I can. Please have look at the attached file and help me.

I am using less 24 hours of data. Its gym validation data of few hours only.  

[meta_GYM.csv.zip](https://github.com/wadpac/GGIR/files/5374002/meta_GYM.csv.zip)
[GYM.zip](https://github.com/wadpac/GGIR/files/5374007/GYM.zip)
[Error.zip](https://github.com/wadpac/GGIR/files/5374017/Error.zip)

andrea-narayan, 2022-09-20:
Did this get solved?
I'm having the same problem

vincentvanhees, 2022-09-21:
> Did this get solved?
> I'm having the same problem

Please provide minimal reproducible example as I suggested above.

q-w-a, 2022-10-15:
Greetings,

I am also having this issue. No error message is produced, but no output for part 5 is generated.

My code is below:

`

library(GGIR)
GGIR::GGIR(datadir=paste0(here::here(), "/data/test"),
           outputdir=paste0(here::here(), "/data/ggir_results"),
           minloadcrit = 6, 
           minimum_MM_length.part5 = 6, 
           dofirstpage =FALSE, 
           do.parallel = FALSE, 
           strategy = 1, 
           includedaycrit =6,
           overwrite = TRUE
)
`

And I am attaching two example files. Like [ksarafrazi](https://github.com/ksarafrazi), these files don't have multiple days of data, so the issue seems to be related to that. Note the files will need to be unzipped before use.
[001 (2019-10-18)RAW.csv.zip](https://github.com/wadpac/GGIR/files/9791857/001.2019-10-18.RAW.csv.zip)
[002 (2019-10-30)RAW.csv.zip](https://github.com/wadpac/GGIR/files/9791858/002.2019-10-30.RAW.csv.zip)

tarensanders, 2022-10-17:
@q-w-a that seems like a different issue because you don't get the error. I can confirm that there is no part 5 metadata with your files.

@vincentvanhees can you clarify if GGIR does not provide part 5 data for single night files? Here's an MRE using the same test that is used to check that part 3 can do single night files:

``` r
# Setup some temp folders
temp_dir <- normalizePath(tempdir())
data_dir <- file.path(temp_dir, "test")
outdir <- file.path(temp_dir, "out")
dir.create(data_dir)
dir.create(outdir)
# create the same test file as part 3 test
GGIR:::create_test_acc_csv(Nmin = 2000, storagelocation = data_dir)
GGIR::GGIR( # these setting match the part 3 test
  mode = 1:5, datadir = data_dir, outputdir = outdir, studyname = "test",
  f0 = 1, f1 = 1, do.report = c(2,4,5), visualreport = FALSE, viewingwindow = 1,
  overwrite = TRUE, do.parallel = FALSE, minimumFileSizeMB = 0,
  includedaycrit = 1, includedaycrit.part5 = 1 / 24, minimum_MM_length.part5 = 1
)
#> 
#>    GGIR version: 2.8.2
#> 
#>    Do not forget to cite GGIR in your publications via a version number and
#>    Migueles et al. 2019 JMPB. doi: 10.1123/jmpb.2018-0063. 
#>    See also: https://cran.r-project.org/package=GGIR/vignettes/GGIR.html#citing-ggir
#> 
#>    To make your research reproducible and interpretable always report:
#>      (1) Accelerometer brand and product name
#>      (2) How you configured the accelerometer
#>      (3) Study protocol and wear instructions given to the participants
#>      (4) GGIR version
#>      (5) How GGIR was used: Share the config.csv file or your R script.
#>      (6) How you post-processed / cleaned GGIR output
#>      (7) How reported outcomes relate to the specific variable names in GGIR
#> ________________________________________________________________________________
#>  Part 1 
#> 
#> Checking that user has read access permission for all files in data directory: Yes
#> Checking that user has write access permission for directory specified by argument outputdir: Yes
#> 1 
#> P1 file 1
#> 
#> Investigate calibration of the sensors with function g.calibrate:
#> 
#> Loading chunk: 1 2 3 4
#> Extract signal features (metrics) with the g.getmeta function:
#> 
#> Loading chunk: 1 2 3
#> Save .RData-file with: calibration report, file inspection report and all signal features...
#> 
#> ________________________________________________________________________________
#>  Part 2 
#> 1 
#> ________________________________________________________________________________
#>  Part 3 
#> 1
#> 
#> ________________________________________________________________________________
#>  Part 4 
#>  1
#> 
#> ________________________________________________________________________________
#>  Part 5 
#> 1 
#> ________________________________________________________________________________
#>  Report part 2 
#>  1
#> 
#> ________________________________________________________________________________
#>  Report part 4 
#>  loading all the milestone data from part 4 this can take a few minutes
#> 
#> ________________________________________________________________________________
#>  Report part 5
part3_filepath <- file.path(
  outdir, "output_test", "meta", "ms3.out",
  paste0("123A_testaccfile.csv.RData")
)
part5_filepath <- file.path(
  outdir, "output_test", "meta", "ms5.out",
  paste0("123A_testaccfile.csv.RData")
)
file.exists(part3_filepath) # the part 3 file exists
#> [1] TRUE
file.exists(part5_filepath) # but not the part 5
#> [1] FALSE
```

<sup>Created on 2022-10-17 with [reprex v2.0.2](https://reprex.tidyverse.org)</sup>

There is a check in part 5 to make sure the file was processed for sleep:
https://github.com/wadpac/GGIR/blob/f4e8b91648d5baa29fe77b9217844c0d624a9174/R/g.part5.R#L165

But this check would fail if `nrows(summarysleep) == 1`, which is one night of sleep?

vincentvanhees, 2022-10-17:
We need at least 2 consecutive nights of sleep to say anything meaningful about time spent in a day. This applies regardless of whether we define the day from midnight to midnight (timewindow = MM) or from wakingup-wakingup (timewindow =WW).

Without sleep detected in the nights on which a day starts and ends it will be impossible to know how to handle lack of movement during that day because it could be part of sleep or it could be inactivity (sedentary behaviour). The primary aim of GGIR part 5 is to help segment the days in sleep, level of physical activity intensity.

andrea-narayan, 2022-10-20:
I've managed to extract data from other files so I'm thinking something may
be wrong with the one that is getting this error (attached below)
Thanks for your help and patience!
This is what I'm using:
 Week 1CBDINS040_left wrist_048331_2022-04-19 12...
<https://drive.google.com/file/d/1EcvwsQyRxsahpdBFMJvVFSeik3DRSmdQ/view?usp=drive_web>

library(GGIR)
g.shell.GGIR(
  mode=c(1,2,3,4,5),
  datadir="W_data",
  outputdir="W_output",
  do.report=c(2,4,5),

  #=====================
  # Part 2
  #=====================
  strategy = 1,
  hrs.del.start = 0,          hrs.del.end = 0,
  maxdur = 9,                 includedaycrit = 16,
  qwindow=c(0,24),
  mvpathreshold =c(100),
  bout.metric = 4,
  excludefirstlast = FALSE,
  includenightcrit = 16,
  #=====================
  # Part 3 + 4
  #=====================
  def.noc.sleep = 1,
  outliers.only = TRUE,
  criterror = 4,
  do.visual = TRUE,
  #=====================
  # Part 5
  #=====================
  threshold.lig = c(30), threshold.mod = c(100),  threshold.vig = c(400),
  boutcriter = 0.8,      boutcriter.in = 0.9,     boutcriter.lig = 0.8,
  boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig = c(1,10),
  boutdur.mvpa = c(1),
  includedaycrit.part5 = 2/3,
  #=====================
  # Visual report
  #=====================
  timewindow = c("WW"),
  visualreport=TRUE)

andrea-narayan, 2022-10-20:
Something must have happened to the actigraphy watch resulting in <2 nights
of data from the looks of it.