wadpac/GGIR #213: Timestamp (start and end time) issue

wwbyun, 2019-09-05:
Hello Vincent,

I have an issue in processing gt3x files, so hope you can provide a solution for it.
After running part 1 and part 2 of g.shell.GGIR, start and end times created under timestamp don’t match with the original start and end times recorded in gt3x files. We’ve observed that the start time is always set in 15-min windows ( 00, 15, 30, or 45 minutes) and the end time is just randomly different from the original end time. 

I provide the code that I used for your information below.

Thank you!

g.shell.GGIR(mode=mode,datadir=datadir,outputdir=outputdir,f0=f0,f1=f1,
             #part 1 related - calibrate, clean and condense
             chunksize=chunksize,
             windowsizes = c(1,900,3600),
             do.cal=do.cal, 
             #part 2 related - impute, summarise
             strategy = strategy, 
             hrs.del.start = hrs.del.start,
             hrs.del.end = hrs.del.end,
             maxdur = maxdur,
             includedaycrit = includedaycrit,
             qwindow=c(0,24), qlevels = qlevels, ilevels = ilevels,
             mvpathreshold = mvpathreshold,  boutcriter = boutcriter,
             idloc=idloc,do.imp=do.imp,
             print.filename=T)

vincentvanhees, 2019-09-12:
GGIR aggregates data per integer numbers of the window sizes specified relative to the whole minute or hour.So, a fifteen minute block starts at minute 00, 15, 30, 45 of the hours. A five second epoch starts at second 5, 10, 15, 20, ...55 of the hour.

At the end of the measurement data can be lost when:
- It does not fit into the last full epoch
- When the last chunk of data loaded is not more than 1 hour, I believe.

Can you calculate the differences you observed and report them here as well as the accelerometer brand and data format you are working with?
Best, Vincent

vincentvanhees, 2019-09-13:
Thanks this is in line with what I would expect.

The argument windowsizes is a vector with three values specifying the epoch lenght for short, medium and long epochs. The latter two are used for non-wear detection and I would not make them shorter than 300 seconds (5 minutes). The first value in the vector you can set to 1 second if you want.

wwbyun, 2019-09-13:
Can I make it 1 second epoch if I don't need non-weartime detection? Or, 5 minute windowsize would be the shortest possible?

vincentvanhees, 2019-09-13:
If you do not need non-wear detection then make the second and third value larger, because the shorter the window the more time consuming the computation.
So, stick with c(5, 900, 3600) or go to for example c(5, 3600, 3600).

The first value you can change to 1, but notice that sleep detection has not been evaluated yet for this epoch length.

wwbyun, 2019-09-13:
Sorry for lingering questions, but one more question. So, there is no way to make the start time matched with the original start time for example 2:08? In other words, is there any way to aggregate data  per 1 minute, which will make the start time at any given minute not 15 minute block?

The issue we really try to solve is non-matching start and end time between two files (GGIR and GT3x).

vincentvanhees, 2019-09-13:
I implemented this intentionally to facilitate comparisons between files and external data sources:
A short epoch that starts a few seconds before the minute or a long epoch that starts a few minutes before the hour will be hard to compare against any external data source or against a different accelerometer recording that started at a different timestamp. Further, data analysis are typically relative to a 24 clock. Epochs that overlap with midnights would then become ambiguous.

There is no option in GGIR at the moment to turn this off.

wwbyun, 2019-09-13:
Okay. It seems that the only option for matching the start and end time is by trimming data in the other file (gt3x/csv) at start and end, right?

vincentvanhees, 2019-09-13:
I think that would be most direct approach.
Other options are:
- Convince the developer of the other software you are working with to do the same as what GGIR does.
- Create an adjusted version of GGIR that can export the data relative to the exact starting time of the measurement. Either do this yourself or hire me or someone else to do it.

wwbyun, 2019-09-13:
I'm curious what's the process of hiring you to do it?

vincentvanhees, 2019-09-13:
I am self-employed, so the best way is to send me a message either via the [website](https://www.movementdata.nl/) or [directly](vincentvanhees@gmail.com) with a reference to this issue, and an indication of how soon you want this to be done. Thanks

wwbyun, 2019-09-13:
Just sent you an email.

vincentvanhees, 2019-10-06:
I will now close this issue. 
I am preparing release 1.10-6.
Following our conversation this version of GGIR will facilitate a 1 minute resolution for non-wear detection, which will enable starting the output at the first full minute in the recording.

closed 2019-10-06 by vincentvanhees