wadpac/GGIR #216: foreach when developing the package expects GGIR to be installed

vincentvanhees, 2019-09-09:
Following [message](https://groups.google.com/d/msg/rpackageggir/ns93tAnMfTQ/3AQA1pAZAQAJ) from Matt in the google group:

In the foreach calls we use argument .packages = 'GGIR',  to specify that GGIR needs to be used.
When developing the GGIR code, this is not practical because we do not want to rebuild the package with every update of the code.

To do: investigate whether there is a better solution.

vincentvanhees, 2019-09-13:
Whether it is installed may not be relevant, because in development mode we may like the package to source the actual R code and not to load the installed GGIR.

We can check whether GGIR is loaded with "GGIR" %in% .packages()
If TRUE, then we know that we are inside an installed package and we can pass on 'GGIR' to the foreach call.
If FALSE, then we know we are in development mode and we can pass on the GGIR functions that are available to the foreach loop directly with .export=c("g.analyse","g.getmeta", etcetera)

I am going to try to see whether this works.

vincentvanhees, 2019-09-13:
It works. I am ow able to both switch from sourcing functions directly and to using functions available from an installed version of GGIR. I have tested with g.part2 and will now do the same for the other 4 parts. Next, I will push these to a development branch, and will sent you a review request when I am done.

Maybe good to clarify that when I am working on the code I usually have the following lines in an R script to source all the files in package directly. I have no idea how other R package developers do this, but for me it works. It avoid having to build the package every time. I am assuming you do something similar.

```
dirR = "/home/vincent/GGIR/R"
ffnames = dir(dirR) # creating list of filenames to source
for (i in 1:length(ffnames)) {
  source(paste(dirR,"/",ffnames[i],sep="")) #source functions
}
library("Rcpp") # also make the cpp functions available (only relevant for .cwa data files)
pathR = "/home/vincent/GGIR"
sourceCpp(paste0(pathR,"/src/numUnpack.cpp"))
sourceCpp(paste0(pathR,"/src/resample.cpp"))

```

m-patterson, 2019-09-16:
yeah, I do the same thing, my sourcing script looks like this, as I don't generally use .cwa files:

all_files = dir('~/R/GGIR/R')
for (iFile in all_files) {
  file_loc = paste0('~/R/GGIR/R/',iFile)
  source(file_loc)
}

vincentvanhees, 2019-09-16:
Thanks Matt, did you see my review request [here](https://github.com/wadpac/GGIR/pull/222)?

m-patterson, 2019-09-16:
Yes, I’m having a look at it now

closed 2019-09-17 by vincentvanhees